RobotLocomotion/drake-external-examples #6: CMake Project Example

stonier, 2017-06-15:
To put under CI and be instructive about how to pull in an installed Drake from a CMake project.

**Travis Example**

This is the one @RussTedrake would like to show (I agree). The point is to have a lightweight Travis build using a hosted tarball of the installation. We'd probably need something like:

* Platform target for this example: Xenial
* [ ] Jenkins builds and hosts a tarball snapshot of drake (nightly?)
* [ ] Working CMake project that does a `find_package(drake)` and compiles a small application using a method in drake.
* [ ] Travis instructions for pulling the hosted snapshot and then building this cmake project.
* [ ] Find success on travis!
* [ ] Enable an automatic periodic run on travis? 
* A nice README.md ...

stonier, 2017-06-15:
How does the above sound?

RussTedrake, 2017-06-15:
Linking to https://github.com/robotlocomotion/drake/issues/6149 (for those with access).

The above sounds great.  Almost any of the simple executables in the examples directory would be appropriate here.  For instance, implementing examples/simple_continuous_time_system again, but using drake as an external.  Perhaps a similarly simple example using the optimization library, too.  Goal is to document/prototype how we think external projects should use drake.

stonier, 2017-07-13:
@jamiesnape we need some concrete examples to validate it. As Russ mentioned, examples/simple_continuous_time_system](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/simple_continuous_time_system.cc) is a good example and I think sufficient for a first pass.

jamiesnape, 2017-07-13:
Sounds good.

stonier, 2017-07-13:
Let's drop that example in all the drake examples (i.e. bazel and cmake). We should also drop the hello world code then just to keep it simple.

jamiesnape, 2017-07-13:
Cool, working on it now.

jamiesnape, 2017-07-13:
Note that the licenses are currently different for this repo and drake and hence `simple_continuous_time_system.cc`.

stonier, 2017-07-13:
Oversight, let's sync the license here with Drake's.

stonier, 2017-07-13:
Some update. We cannot do Travis, it has no support for Ubuntu/Xenial. CircleCI is another option, but 2.0 with Xenial support only released yesterday. Likely to be bumpy. There are other options that may be feasible, e.g. docker container builds, but these are likely to be sufficiently complex to defeat the purpose.

For now we will just drop in a simple bash script build on Jenkins. Bash scripts in the project examples.

stonier, 2017-07-13:
Task List for the 20170713-14 sprint:

Target Platform: Ubuntu/Xenial

* [ ] Jenkins builds and hosts a tarball snapshot of drake (nightly?)(where?)
* [ ] CMake `find_package` project example copies and compiles [examples/simple_continuous_time_system](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/simple_continuous_time_system.cc)
* [ ] CMake `ExternalProject_Add` project example copies and compiles [examples/simple_continuous_time_system](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/simple_continuous_time_system.cc)
* [ ] Jenkins bash scripts inside both cmake project examples
* [ ] Jenkins jobs on our CI server that builds each cmake project examples (nightly?)
* [ ] A nice README.md in each project example

RussTedrake, 2017-07-14:
@rdeits, @tkoolen -- any ideas for getting around lack of ubuntu support?

jwnimmer-tri, 2017-07-14:
Travis-CI says to use Docker if you want a Xenial build environment https://github.com/travis-ci/travis-ci/issues/5821#issuecomment-202028783.  I think that's a fine approach for a Xenial flavor of the a prebuilt-Drake-in-other-CI example.

jamiesnape, 2017-07-14:
I think CircleCI 2.0 would work best for us at this stage.

rdeits, 2017-07-14:
I set up xenial testing with Docker on Travis for Director: https://github.com/RobotLocomotion/director/blob/master/.travis.yml and it was actually pretty easy.

stonier, 2017-07-26:
Most jobs covered in PR #8. Will follow up on the remaining in individual issues.

closed 2017-07-26 by stonier