RobotLocomotion/drake-external-examples #9: Python Project Example

stonier, 2017-07-13:
We should probably also exemplify how to pull in an installed drake and use it inside a python project.

@sammy-tri are there pure python only users of drake?

sammy-tri, 2017-07-13:
There are parts of director which interact with parts of drake only through the python interface...  but I can't think of an example of a project which only uses drake through python off hand.

dmsj, 2017-09-22:
FWIW, at Draper we would like a pure python workflow for working with drake and pytorch for learning manipulation strategies. 

I would really like a 'setup.py' to install drake for use with a specific python installation. Even better would be a conda package ;-) 

I think there are quite a few applications which would be enabled with a really simple, easy-to-install python interface to Drake.

stonier, 2017-09-28:
Be interested to learn how you wish to have those two working together. @dmsj are you on slack?

dmsj, 2017-09-28:
@stonier yup - i'm on Drake Developers. I'll message you there. 

re: use-case - To begin with, we would like to be able to simulate a dynamic system (with contact) and control it with a learned policy implemented in pytorch. We would use the sim for training as well as evaluation.

RussTedrake, 2017-09-28:
cc @EricCousineau-TRI -- he's about to prioritize cleaning up and expanding the python bindings for drake (including the systems framework).

EricCousineau-TRI, 2018-02-09:
Closed via #85.

closed 2018-02-09 by EricCousineau-TRI