RobotLocomotion/drake-external-examples #11: create at least one example that sources a resource (e.g. urdf file) from drake

RussTedrake, 2017-07-24:
now that we've seen issues with finding resources (with FindResource) from external repos, I think it's important that we capture that in our example here.  @siyuanfeng-tri was the one who battled through it most recently.

stonier, 2017-07-26:
@siyuanfeng-tri would the [double pendulum demo](https://github.com/RobotLocomotion/drake/tree/master/drake/examples/double_pendulum) be an appropriate candidate to make an example of here?

siyuanfeng-tri, 2017-07-26:
@stonier yeah sure. Any model would fine. Last time I tried this, I was having trouble loading a abs path urdf into an RBT. FindResource complains that the file is not within drake basically.

jwnimmer-tri, 2017-07-26:
Yes, never pass an absolute path to [FindResource](http://drake.mit.edu/doxygen_cxx/namespacedrake.html#a1becb3d158eac50e74aa9f1921c3a2b3) -- on the tin, it says to use relative paths.

jwnimmer-tri, 2017-07-26:
(If you have an absolute path, then you can just call `AddModelInstanceFromUrdfFile` (or similar) directly.)

stonier, 2017-11-07:
Resolved by #52

closed 2017-11-07 by stonier