RobotLocomotion/drake-external-examples #18: Binaries do not include data files

stonier, 2017-07-26:
This is a pre-requisite of having #11 working and under test.

jamiesnape, 2017-07-26:
FYI @fbudin69500.

jamiesnape, 2017-07-26:
Do we have a list of data files needed yet? Can we rationalize their locations in Drake to make it more obvious which are part of the API and can be used by external projects, i.e., need installing?

jwnimmer-tri, 2017-07-26:
This sounds like a Drake issue, not a Shambhala issue?

jamiesnape, 2017-07-26:
Yes, though I would not object to having cross referenced issues on both in this instance.

stonier, 2017-07-26:
Detailed issue tracking for this over in https://github.com/RobotLocomotion/drake/issues/6668

stonier, 2017-08-21:
Examples and manipulation data is now in via: https://github.com/RobotLocomotion/drake/pull/6671

closed 2017-08-21 by stonier