RobotLocomotion/drake-external-examples #23: separate repos to support a "clone and go" workflow?

RussTedrake, 2017-07-28:
@jamiesnape -- can't tell you how excited i am to see the progress in this repo.  it's looking great and will be a huge contribution.  thank you!

i have a question/proposal that i asked @stonier and @jwnimmer-tri yesterday:  what if we were to zap this repo, and instead have separate repos titled `drake-bazel-external`, `drake-cmake-external`, and `drake-cmake-installed`?  those directories would have the travis.yml file in their root (the right spot), etc, and would therefore make it even simpler for a user to get started.  they could essential clone the repo, push to their new repo, and be on travis/circle immediately.  I admit it's only moderately harder to clone this repo, then copy only one subdirectory, move the travis.yml file, etc, but I think that difference gets bigger for novice users.  

The downside @stonier articulated is that some of the scripts are being shared between the subfolders.  But I would prefer to increase the cost slightly of us maintaining 3 repos if it can decrease the cost to a novice user.  

Side benefit: the repo names will make sense to everyone.  ;-)

What do you think?

jamiesnape, 2017-07-28:
We already duplicate a lot of code, so I really would not have a huge issue.

stonier, 2017-07-31:
@RussTedrake after some thinking, still not really keen on this. There will be additional use cases that qualify as very important first steps for getting started. e.g. pip package (pybindings), catkin package.  It's much nicer seeing the options available than having available a simplistic (and ultimately annoying because of the forked link) fork. Our users really ought not to be quite that naive - if they are, they're certainly not going to dive into c++ drake programming.

Ultimately long term I'd hope we don't even need shambhala. i.e. it becomes conventional enough not to need esoteric demonstrations of how to get started with it. Apart from the bundled dependencies and lack of a .deb, the CMake installed project for Xenial is already nigh on conventional - it's almost a purely CMake - Travis tutorial.

RussTedrake, 2017-07-31:
Ok.  Let's start this way and see how it goes for users?