RobotLocomotion/drake-external-examples #31: Pull the 'Underactuated' textbook from Drake into this repository

stonier, 2017-08-22:
Russ mentioned that it might be worth pulling the underactuated textbook into this repository from Drake itself.

If we start doing this, then Shambhala starts to become as it's name suggests - the portal to getting started with Drake, not just an exemplary instance of how to build a package using Drake.

@RussTedrake this might be a good question to pose to the first Drake Developers meeting?

RussTedrake, 2017-08-22:
Sorry - i meant that the textbook should pull in drake as an external, an instance of the capability demonstrated here.  I don't think the textbook shojld actually be in this repo.

stonier, 2017-08-23:
Aha, I should have gone investigating first instead of working on some assumptions.

And yes, having the textbook follow the usage pattern here will have them spend more time learning about underactuated systems rather than dealing with complicated build/sync issues.

Did you want to drop the textbook submodule from Drake itself?

RussTedrake, 2017-08-23:
That's exactly right.  I mentioned it to @jamiesnape once before, but we can pull the textbook external from drake at any time.

closed 2017-08-23 by stonier