RobotLocomotion/drake-external-examples #38: Add examples using PCL

EricCousineau-TRI, 2017-09-28:
Since PCL is a very common library, it should be incorporated into some downstream project as an example.

The steps I can think of at the moment:

- [ ] Incorporate a custom-built Drake-compatible version of PCL that can be used in Bazel (as an archive) - see #37 for a prototype. Focus is first on Xenial, possibly on Mac thereafter.
- [ ] Incorporate a custom-built Drake-compatible version of PCL within the CMake external (preferrably `cmake_install`, using the existing archive that Bazel consumes)
- [ ] Once Drake is compatible with a more widely distributed ROS- / Ubuntu-package, incorporate this into CI.

stonier, 2017-10-23:
@EricCousineau-TRI is there any reason you do not directly use the system version (1.7.x)?

For the CMake build, it has Eigen conflicts (see [drake_pcl](https://github.com/stonier/drake_pcl/blob/devel/CMakeLists.txt). Working on finding a way to resolve that so it can co-exist with other ROS packages.

stonier, 2017-10-23:
Note: as indicated in the PR, official bazel external support has been dropped for now. I'd kick back on getting it into Drake examples unless you start using it for core library methods as it blows up the dependency tree and maintenance thereof (i.e. Drake can't be examples from N user groups and M*N use cases on how to build things).

EricCousineau-TRI, 2017-10-24:
>  is there any reason you do not directly use the system version (1.7.x)?

I had not used the system version due to the difference in the Eigen build and linking options, and I wanted to avoid those issues at least at the beginning, and then transition to the system-distributed version (Step 3 in the checklist above).

>  I'd kick back on getting it into Drake examples unless you start using it for core library methods

Sounds good. At some point in the future, I will want it in core functionality, but will go down that road once it comes closer.

stonier, 2017-11-06:
@nuclearsandwich has two candidate solutions:

* [CMake Overrides](https://github.com/osrf/drake_pcl/blob/prototype-override-only/CMakeLists.txt#L24) which tricks the Xenial PCLConfig.cmake into finding it by setting `EIGEN_FOUND` and `EIGEN_INCLUDE_DIRS` (as opposed to Eigen3...).
* [Custom Find CMake Module](https://github.com/osrf/drake_pcl/tree/prototype-pcl-config)

The first option is the most convenient, but does defy CMake convention for Eigen. The second possible, but brittle seeing as it's replicating alot of logic in the system `PCLConfig.cmake`. A third option might be to bake our own `find_pcl` macro which isolates the abuse of the `EIGEN` variables.

@jamiesnape recommendations?
@siyuanfeng-tri FYI

jamiesnape, 2017-11-06:
Write a simple `FindPCL` module that calls `find_package(Eigen)`, sets `EIGEN_FOUND`, etc. if found, and then calls `find_package(PCL CONFIG)`.

jamiesnape, 2017-11-06:
See also `CONFIG_MODE` of `find_package_handle_standard_args`.

jamiesnape, 2017-11-06:
You may need `get_target_property` to get the include directory from the target.

stonier, 2017-11-06:
Thanks @jamiesnape

stonier, 2017-11-07:
Yak shaving problem post Eigen resolution with vtk:

```
[100%] Linking CXX executable my_test
/usr/bin/ld: cannot find -lvtkftgl
/usr/bin/ld: cannot find -lvtkIOParallelNetCDF
/usr/bin/ld: cannot find -lvtkParallelMPI
/usr/bin/ld: cannot find -lvtkRenderingOpenGL
/usr/bin/ld: cannot find -lvtkLocalExample
/usr/bin/ld: cannot find -lvtkTestingGenericBridge
/usr/bin/ld: cannot find -lvtkViewsQt
/usr/bin/ld: cannot find -lvtkxdmf2
/usr/bin/ld: cannot find -lvtkFiltersReebGraph
/usr/bin/ld: cannot find -lvtkIOXdmf2
/usr/bin/ld: cannot find -lvtkRenderingContextOpenGL
/usr/bin/ld: cannot find -lvtkGUISupportQtOpenGL
/usr/bin/ld: cannot find -lvtkIOParallelLSDyna
/usr/bin/ld: cannot find -lvtkFiltersParallelGeometry
/usr/bin/ld: cannot find -lvtkGUISupportQtWebkit
/usr/bin/ld: cannot find -lvtkRenderingVolumeOpenGL
/usr/bin/ld: cannot find -lvtkIOPostgreSQL
/usr/bin/ld: cannot find -lvtkWrappingJava
/usr/bin/ld: cannot find -lvtkFiltersParallelFlowPaths
/usr/bin/ld: cannot find -lvtkFiltersParallelStatistics
/usr/bin/ld: cannot find -lvtkRenderingParallelLIC
/usr/bin/ld: cannot find -lvtkRenderingLIC
/usr/bin/ld: cannot find -lvtkIOParallelExodus
/usr/bin/ld: cannot find -lvtkRenderingQt
/usr/bin/ld: cannot find -lvtkGUISupportQtSQL
/usr/bin/ld: cannot find -lvtkRenderingFreeTypeOpenGL
/usr/bin/ld: cannot find -lvtkInfovisBoostGraphAlgorithms
/usr/bin/ld: cannot find -lvtkRenderingGL2PS
/usr/bin/ld: cannot find -lvtkIOGeoJSON
/usr/bin/ld: cannot find -lvtkViewsGeovis
/usr/bin/ld: cannot find -lvtkTestingIOSQL
/usr/bin/ld: cannot find -lvtkPythonInterpreter
/usr/bin/ld: cannot find -lvtkIOODBC
/usr/bin/ld: cannot find -lvtkIOMySQL
/usr/bin/ld: cannot find -lvtkRenderingMatplotlib
/usr/bin/ld: cannot find -lvtkFiltersParallelMPI
/usr/bin/ld: cannot find -lvtkTestingRendering
/usr/bin/ld: cannot find -lvtkIOMPIParallel
/usr/bin/ld: cannot find -lvtkParallelMPI4Py
/usr/bin/ld: cannot find -lvtkIOVPIC
/usr/bin/ld: cannot find -lVPIC
/usr/bin/ld: cannot find -lvtkRenderingParallel
/usr/bin/ld: cannot find -lvtkRenderingFreeTypeFontConfig
/usr/bin/ld: cannot find -lvtkIOFFMPEG
/usr/bin/ld: cannot find -lvtkIOMPIImage
/usr/bin/ld: cannot find -lvtkIOGDAL
collect2: error: ld returned 1 exit status
CMakeFiles/my_test.dir/build.make:319: recipe for target 'my_test' failed
make[2]: *** [my_test] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/my_test.dir/all' failed
make[1]: *** [CMakeFiles/my_test.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

i.e. for some reason the drake-pcl combination is not finding system vtk (for pcl). 

@nuclearsandwich is looking into the problem.

jamiesnape, 2017-11-07:
Did you call `find_package(VTK CONFIG REQUIRED)` in `FindPCL`?

jamiesnape, 2017-11-07:
(I don't have a copy of PCL at hand to know if they call `find_dependency(VTK)` in `PCLConfig` as they should.)

stonier, 2017-11-07:
They do not use `CONFIG`. 

`PCLConfig.cmake`:

```
macro(find_VTK)
  if(PCL_ALL_IN_ONE_INSTALLER AND NOT ANDROID)
    set(VTK_DIR "${PCL_ROOT}/3rdParty/VTK/lib/vtk-5.8")
  elseif(NOT VTK_DIR AND NOT ANDROID)
    set(VTK_DIR "/usr/lib/cmake/vtk-6.2")
  endif(PCL_ALL_IN_ONE_INSTALLER AND NOT ANDROID)
  if(NOT ANDROID)
    find_package(VTK ${QUIET_})
    if (VTK_FOUND)
      set(VTK_LIBRARIES "vtkImagingStencil;....
  endif()
endmacro(find_VTK)
```

jamiesnape, 2017-11-07:
And you have `libvtk6-dev` installed?

stonier, 2017-11-07:
System `libpcl-dev` requires `libvtk6-dev`.

jamiesnape, 2017-11-07:
Look in CMakeCache.txt. What is the value of `VTK_DIR`?

stonier, 2017-11-07:
```
VTK_DIR= /opt/drake/lib/cmake/vtk-8.0
```

jamiesnape, 2017-11-07:
Thought so. If you do check with `ldd`, does `libdrake.so` explicitly link VTK? If so, we can remove it from `drake.cps` and this will work again. If not, we have a problem.

stonier, 2017-11-07:
```
ldd /opt/drake/lib/libdrake.so  | grep vtk
        libvtkIOGeometry-8.0.so.1 => /opt/drake/lib/libvtkIOGeometry-8.0.so.1 (0x00007fe76e2db000)
        libvtkIOImage-8.0.so.1 => /opt/drake/lib/libvtkIOImage-8.0.so.1 (0x00007fe76df6b000)
        ...
```

If you remove the explicit linking, what happens should someone try and use a drake method that internally calls vtk8 api? Wouldn't that end up failing at runtime?

jamiesnape, 2017-11-07:
We are not removing the explicit linking within libdrake.so, we are relying on it to handle that case. The entry in the CPS was to handle the case that there was not any explicit linking, but apparently there is now, due to this, I think: https://github.com/RobotLocomotion/drake/blob/bdf86c07394b310a1f07b4e024fa3b84aed43d99/tools/install/libdrake/BUILD.bazel#L109

jamiesnape, 2017-11-07:
Looks like we can also prune TinyXML2 for the same reason. Basically, if you directly link something into a `cc_binary` everything links properly. If you did not mind the duplication, then we could remove all the `Link-Requires` by explicitly linking `libdrake.so` with those dependencies.

jamiesnape, 2017-11-07:
I slightly misspoke. The reason is that they are system dependencies, so looks like we can remove `nlopt` and `zlib` as well.

nuclearsandwich, 2017-11-07:
> Write simple FindEigen3 and FindPCL modules that calls find_package(Eigen), sets EIGEN_FOUND, etc. if found, and then calls find_package(PCL CONFIG). No brittleness, no duplication, works for any Eigen.

Should these find modules be distributed with Drake or should we incorporate them into this example library as a reference for people to copy?

jamiesnape, 2017-11-07:
They should be distributed with Shambhala since Drake does not use PCL directly.

stonier, 2017-11-07:
Shambhala is not something that people need use/install (or at least it isn't now). Many people simply do a `bazel run //:install /opt/drake` and would then like to kickstart their own CMake project that used Drake and PCL. 

If Drake has a problem being compatible with other projects, e.g. PCL, then I think it is most convenient to have the manual installation process drop any required machinery in /opt/drake along with the rest.

jamiesnape, 2017-11-07:
I do not think this is a Drake problem to solve. This is an example of how to use Drake and PCL and people should follow the example. We are basically patching the problems with the PCL build system.

jamiesnape, 2017-11-07:
(We also don't want people doing `bazel run //:install /opt/drake` anymore, we have a pure CMake workflow now.)

stonier, 2017-11-07:
We are patching problems that get discovered because we refrain from using the common system Eigen. I do not want to shift the burden of that choice to users.

Specifically I do not want them to have to maintain a `FindPCL.cmake` that is needed only for Drake-PCL projects if we can do so without any trouble. We similarly bundle third party dependencies with Drake so that users don't have to separately fetch these. Unless there's a good reason not to, then I'd like to support a simple philosophy of "here's a bundle that includes Drake and esoterica required to use Drake" especially if they are directly required by our internal group. 

At least until we have better machinery to help with a distributed set of Drake requirements.

stonier, 2017-11-07:
> (We also don't want people doing bazel run //:install /opt/drake anymore, we have a pure CMake workflow now.)

This is a current workflow for installing drake with proprietary solvers and is the only way to develop if you are working on both Drake and your downstream CMake project(s) simultaneously.

jamiesnape, 2017-11-07:
I think the solution, since we are stuck with a Bazel built version, is therefore to ensure that it provides everything that the real Eigen has. Therefore, unfortunately, that means a pkg-config file.

EricCousineau-TRI, 2019-02-08:
I think that this issue is stale; this should be tracked by #91.

closed 2019-02-08 by EricCousineau-TRI