RobotLocomotion/drake-external-examples #39: Consider switching to BSD license ASAP

jwnimmer-tri, 2017-10-02:
(Forked off from #37.)

While I love the MIT license better than BSD, the fact that Shamb is MIT and Drake is BSD means that when we copy code between them, we are going to have to fiddle with license grant badging in annoying ways.

Given the tightly-coupled relationship between Drake and Shamb, please consider having Shamb change to use the BSD license ASAP, to reduce developer friction.

stonier, 2017-10-02:
It is actually a BSD 3 clause license. The copyright to MIT makes it slightly confusing.

What we should do, as discussed in #37 is make sure each file is licensed w/ a copyright pointing to the appropriate entity (e.g. Toyota Research Institute for contributions from TRI employees).

jwnimmer-tri, 2017-10-02:
Oops.  I think I've made that (MIT vs BSD) mistake in the past on Shamb too.  Sorry!

closed 2017-10-02 by jwnimmer-tri