RobotLocomotion/drake-external-examples #43: Shambhala CI Fail

stonier, 2017-10-04:
Currently failing on the master branch. See [here](https://drake-jenkins.csail.mit.edu/job/RobotLocomotion/job/drake-shambhala/job/master/17/console) - not finding make_unique (c++14 feature).

stonier, 2017-10-04:
This is happening on the externalAddProject build.

stonier, 2017-10-05:
Likely some problem around the arguments going into bazel for the compiler.

https://github.com/RobotLocomotion/drake-shambhala/blob/master/drake_cmake_external/CMakeLists.txt#L48

jwnimmer-tri, 2017-10-10:
I believe the problem is that in addition to passing `--cxxopt=-std=c++1y`, we also need to pass `--host_cxxopt=-std=c++1y`.  Some data-generating tools are compiled for the `host` architecture, instead of `local-opt`, so were not using the option.

closed 2017-10-12 by jamiesnape

stonier, 2017-10-12:
@jwnimmer-tri Thanks for the hint!

PR #44 (2017-10-12): Fix CI build issues
Fixes #43
commit f91388c1 (2017-10-12): Fix CI build issues (#44)

* more rigorous cxx 14 flag settings for a cmake external build of drake

Fixes #43

* dashdash

* add tinyxml2 to the install

* tinyxml2 for mac, libboost-all for trusty and xenial