RobotLocomotion/drake-external-examples #45: CI Fail - stx dependency in the bazel external

stonier, 2017-10-13:
Failure here: https://drake-jenkins.csail.mit.edu/job/RobotLocomotion/job/drake-shambhala/job/PR-35/10/console

```
Analyzing: target //apps:simple_continuous_time_system (10 packages loaded)
ERROR: /home/ubuntu/.cache/bazel/_bazel_ubuntu/4ec442eb70ae159fd7ea51f91a92aa37/external/drake/drake/common/BUILD.bazel:52:1: no such package '@stx//': The repository could not be resolved and referenced by '@drake//drake/common:essential'
```

This popped up due to the change in [Drake #7241](https://github.com/RobotLocomotion/drake/pull/7241/files).

Stx got shifted to utilising a `new_local_repository` approach. This can't be replicated when drake is brought in as an external because the path to those headers can't be retrieved.

It's possibly also not the correct thing to do - the path for a new_local_repository is supposed to absolute according to the [docs](https://docs.bazel.build/versions/master/be/workspace.html#new_local_repository), though obviously the bazel executable doesn't actually back that up. I am guessing that the spirit of this mechanism is for experimenting with dependencies on your local system, not for managing sandboxed sets of files within the project.

stonier, 2017-10-13:
@jwnimmer-tri thoughts?

stonier, 2017-10-13:
This also breaks the cmake external project build.

stonier, 2017-10-13:
Wrong install directory: `install/include/stx/stx/optional.hpp`

jwnimmer-tri, 2017-10-14:
> It's possibly also not the correct thing to do - the path for a new_local_repository is supposed to absolute according to the docs, ...

Good catch.  Filed as RobotLocomotion/drake#7261.

> Stx got shifted to utilising a new_local_repository approach. This can't be replicated when drake is brought in as an external ...

I'll grant you that its obscure, but "can't" is one step too far:
```diff
--- a/drake_bazel_external/WORKSPACE
+++ b/drake_bazel_external/WORKSPACE
index 15d5f75..32cfd6a 100644
@@ -39,20 +39,35 @@
 # Drake
 ################################################################################
 
+_DRAKE_REPO = "RobotLocomotion/drake"
+_DRAKE_COMMIT = "master"
+_DRAKE_SHA256 = None
+# alternatively for a specific tag
+# _DRAKE_COMMIT = "65c4366ea2b63278a286b1e22b8d464d50fbe365",
+# _DRAKE_SHA256 = "899d98485522a7cd5251e50a7a6b8a64e40aff2a3af4951a3f0857fd938cafca",
+
 github_archive(
     name = "drake",
-    repository = "RobotLocomotion/drake",
-    commit = "master",
-    sha256 = None
-    # alternatively for a specific tag
-    # commit = "65c4366ea2b63278a286b1e22b8d464d50fbe365",
-    # sha256 = "899d98485522a7cd5251e50a7a6b8a64e40aff2a3af4951a3f0857fd938cafca",
+    repository = _DRAKE_REPO,
+    commit = _DRAKE_COMMIT,
+    sha256 = _DRAKE_SHA256,
 )
 
 ################################################################################
 # Drake Dependencies
 ################################################################################
 
+new_http_archive(
+    name = "stx",
+    urls = [
+        "https://github.com/%s/archive/%s.tar.gz" % (
+            _DRAKE_REPO, _DRAKE_COMMIT)
+    ],
+    sha256 = _DRAKE_SHA256,
+    strip_prefix = "drake-" + _DRAKE_COMMIT + "/third_party/com_github_tcbrindle_cpp17_headers",
+    build_file = "tools/stx.BUILD",
+)
+
 bitbucket_archive(
     name = "eigen",
     repository = "eigen/eigen",

--- a/drake_bazel_external/tools/stx.BUILD
+++ b/drake_bazel_external/tools/stx.BUILD
new file mode 100644
index 0000000..45cceb5
@@ -0,0 +1,9 @@
+# -*- python -*-
+
+package(default_visibility = ["//visibility:public"])
+
+cc_library(
+    name = "stx",
+    hdrs = ["stx/optional.hpp"],
+    includes = ["."],
+)
```

This mess is consistent with RobotLocomotion/drake#7259 being unfinished.

---

But also, let me take a step back here.  I do not think that Shambhala's `drake_bazel_external` is ripe yet.

To the extent we investigate and fix when it fails CI, we are wasting our time.  To the extent that Drake's users look at it for advice, we are doing them harm.

Its code fails the "don't repeat yourself" sniff test, and it sets users on the wrong path for how to pursue downstream Bazel integration.  It goes against current best practices in the Bazel ecosystem for how to chain Bazel projects together.  Once we finish RobotLocomotion/drake#7259, _then_ we should pursue the `drake_bazel_external` exemplar, and show users how to do it right.

In the meantime, I suggest removing it from `master`.  If that's not tenable, then at least it should have a bigger "The code in this folder is a bad idea and don't copy it!" disclaimer.  The current `**Under Construction**` text doesn't really get that message across strongly enough, in my opinion.

dmsj, 2017-10-15:
@jwnimmer-tri your above code and comment was quite useful! I didn't realize that [drake#7259](https://github.com/RobotLocomotion/drake/issues/7259) was in progress and I was going down a rabbit hole. 

Is the workflow in the `drake-iiwa-driver` bazel build going to be the suggested way forward for external drake apps? We have ended up using many different versions of drake in various parts of the system and would like to start standardizing. 

FWIW - adding your above comment (or linking to it) to the `drake_bazel_external` README would be very useful.

stonier, 2017-10-16:
I'd say that workaround effectively falls into what I consider "can't". It's starting to resort to clever, but obscure/drastic methods which suggests it's now fighting with the system rather than working with it.

Having this here was intended to motivate an exploration in how to do this properly prior to advertising Shambhala. Some efforts were rebuffed and so it lingered. I was about to bump this for resolution (one way or another) as part of a final effort to getting Drake & Shambhala ready and under CI as we already have groups trying various brittle approaches to get this working. In that respect, I am glad that [drake#7259](https://github.com/RobotLocomotion/drake/issues/7259) has now articulated this is a desired goal.

Regardless of whether it will be supported or not, a discerning note concerning the situation for downstream bazel projects would be entirely relevant in a README here since the question is oft asked.

jwnimmer-tri, 2017-10-16:
@dmsj 

> Is the workflow in the drake-iiwa-driver bazel build going to be the suggested way forward for external drake apps? We have ended up using many different versions of drake in various parts of the system and would like to start standardizing.

I don't know `drake-iiwa-driver` much at all.  But from first glance, it seems like it's only grabbing Drake to refer to some of its LCM messages?  I thought that's what https://github.com/RobotLocomotion/lcmtypes was for?  So as the first choice, I guess I would say to stop depending on Drake if possible, in that repository.

In general though, once RobotLocomotion/drake#7259 is ready, then it would be an option to use Bazel in your own code and refer to Drake that way.  Alternatively, Drake will offer a `CMakeLists.txt` that will compile and install Drake in a way that other CMake-based projects can use naturally.  Depending on what your code wants, one of those two ways should be an easy path forward.

jwnimmer-tri, 2017-10-16:
@stonier

> I am glad that drake#7259 has now articulated this is a desired goal.

To be clear, I think we should focus 100% of our efforts on Drake's CMake wrapper to get that support up and running perfectly first.  Only when we are done (or blocked) on that front, should we invest in shipping a supported direct-into-Bazel interface.

stonier, 2017-10-17:
> To be clear, I think we should focus 100% of our efforts on Drake's CMake wrapper to get that support up and running perfectly first. Only when we are done (or blocked) on that front, should we invest in shipping a supported direct-into-Bazel interface.

This is of course, implicit.

closed 2017-10-17 by stonier

jwnimmer-tri, 2017-10-17:
Driving doesn't need or want RobotLocomotion/drake#7259 in any way, shape or form.

stonier, 2017-10-19:
Why would driving not make use of it? It is a downstream bazel project, that at some point may have dependencies that don't already fully intersect with those of Drake's.

jwnimmer-tri, 2017-10-19:
Driving is not allowed to pull dependencies off the internet.  It's entire WORKSPACE is custom, and shares nothing with Drake.

stonier, 2017-10-19:
Noted, thanks.