RobotLocomotion/drake-external-examples #56: macOS build issue with drake_cmake_installed

soonho-tri, 2017-10-30:
```
$ cmake ../ -Ddrake_DIR=~/tmp/drake/lib/cmake/drake

CMake Error at /usr/local/Cellar/cmake/3.9.4_1/share/cmake/Modules/CMakeFindDependencyMacro.cmake:37 (find_package):
  Could not find a package configuration file provided by "VTK" (requested
  version 8.0.1) with any of the following names:

    VTKConfig.cmake
    vtk-config.cmake

  Add the installation prefix of "VTK" to CMAKE_PREFIX_PATH or set "VTK_DIR"
  to a directory containing one of the above files.  If "VTK" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  /Users/soonhok/tmp/drake/lib/cmake/drake/drake-config.cmake:78 (find_dependency)
  CMakeLists.txt:39 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/soonhok/work/sham/drake_cmake_installed/build/CMakeFiles/CMakeOutput.log".
```

Env:
 - macOS 10.13
 - cmake-3.9.4 installed (via homebrew)
 - vtk-8.0.1 installed (via homebrew)

stonier, 2017-10-30:
@soonho-tri - it's a xenial only binary. I guess this means I should make that announcement in LARGE FONT from somewhere.

soonho-tri, 2017-10-30:
Ah.. right. Closing the issue.

jamiesnape, 2017-11-20:
Fixed by https://github.com/RobotLocomotion/drake/pull/7293 and obsoleted by https://github.com/RobotLocomotion/drake/pull/7434.

closed 2017-11-20 by jamiesnape