RobotLocomotion/drake-external-examples #57: Shambhala CI under buildcop purview

stonier, 2017-10-30:
So that it is noticed when things break, at least nightly.

jwnimmer-tri, 2017-11-01:
A question came up on slack as to what the buildcop should do when this breaks.  This was my suggestion:

> Buildcop instructions get updated to say that when Shambhala's entry in `Foo Production` is red, the buildcop opens a Shambhala issue, CCing `@stonier` on it.  And then don't worry about a revert, or fix-foward, or etc.

Until we teach Drake developers what Shambhala is, enable them to contribute to it, and have it under pre-merge testing in Drake CI, I think that's the best we can do.  Right now, its a relatively specialized project that most buildcops and Drake Developers don't know what to do with.

EricCousineau-TRI, 2019-02-07:
Action item to close this: Add `drake-shambhala` to Drake build cop instructions.

EricCousineau-TRI, 2019-03-28:
Given that we have this repo in the build cop build, that should be sufficient traceability.

Should apply Jeremy's patch, then PR.

closed 2019-04-02 by EricCousineau-TRI