RobotLocomotion/drake-external-examples #58: Shambhala CI as a Drake PR hook

stonier, 2017-10-30:
Or some form of it - the external add project would probably be sufficient.

jwnimmer-tri, 2020-09-23:
My worldview is that jobs that fail nightly get promoted to continuous, and jobs that fail continuous get promoted to pre-merge.  Jobs that don't fail can slide back to be run less frequently.

My estimate is that drake-external-examples nightly build fails around once per month, and about half of those are infrastructure issues unrelated to Drake commits.

Therefore, I think that missing this pre-merge integration job is relatively painless, whereas adding it would be somewhat painful.

@jamiesnape If you agree, then perhaps we can just close this request as "not worth it anytime soon" and reopen if it becomes acutely painful.

jamiesnape, 2020-09-23:
I agree. To be honest, I am not sure it would ever be worth it, given the scope of this repo.

jwnimmer-tri, 2020-09-23:
I might be convenient to have a bot command to opt-in to it when changing Drake's build system or prereqs, but yeah probably never as pre-merge by default.

closed 2020-09-23 by jwnimmer-tri