RobotLocomotion/drake-external-examples #69: Need help with superbuild that consumes drake as external.

siyuanfeng-tri, 2017-12-11:
Hi guys,
So here is the project I am working on. https://github.com/siyuanfeng-tri/rand_obj_picking
This repo has 4 main sub components and a executable that depends on them. The particular problem I am running into is during linking for that application. 

As is, this should build just fine. There is a short readme explaining how to make. To illustrate the problem, uncomment https://github.com/siyuanfeng-tri/rand_obj_picking/blob/master/src/test_robot_bridge.cc#L4 and https://github.com/siyuanfeng-tri/rand_obj_picking/blob/master/src/test_robot_bridge.cc#L42
You should see something like 
```[ 95%] Performing build step for 'clutter_proj'
Scanning dependencies of target test_robot_bridge
[ 50%] Building CXX object CMakeFiles/test_robot_bridge.dir/test_robot_bridge.cc.o
[100%] Linking CXX executable test_robot_bridge
/usr/bin/c++   -g -O2 -Wall     CMakeFiles/test_robot_bridge.dir/test_robot_bridge.cc.o  -o test_robot_bridge  -L/home/sfeng/code/rand_obj_picking/build/lib -lperception -lgrasp_gen -lrobot_bridge /home/sfeng/drake_bazel_build/lib/libdrake.so /home/sfeng/drake_bazel_build/lib/libBulletCollision.so /home/sfeng/drake_bazel_build/lib/libLinearMath.so /home/sfeng/drake_bazel_build/lib/libfcl.so /home/sfeng/drake_bazel_build/lib/libccd.so /home/sfeng/drake_bazel_build/lib/liboctomap.so /home/sfeng/drake_bazel_build/lib/libignition_rndf.so /home/sfeng/drake_bazel_build/lib/liblcm.so -lglib-2.0 -lpthread /home/sfeng/drake_bazel_build/lib/libprotobuf.so /home/sfeng/drake_bazel_build/lib/libyaml_cpp.so /home/sfeng/drake_bazel_build/lib/libfmt.so /home/sfeng/drake_bazel_build/lib/libscsdir.so /home/sfeng/drake_bazel_build/lib/libsdformat.so /home/sfeng/drake_bazel_build/lib/libignition_math.so -ltinyxml /home/sfeng/drake_bazel_build/lib/libtinyobjloader.so -ltinyxml2 -Wl,-rpath,/home/sfeng/code/rand_obj_picking/build/lib:/home/sfeng/drake_bazel_build/lib: 
/usr/bin/ld: CMakeFiles/test_robot_bridge.dir/test_robot_bridge.cc.o: undefined reference to symbol '_ZN20vtkDebugLeaksManagerD1Ev'
/home/sfeng/drake_bazel_build/lib/libvtkCommonCore-8.0.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/test_robot_bridge.dir/build.make:112: recipe for target 'test_robot_bridge' failed
make[6]: *** [test_robot_bridge] Error 1
CMakeFiles/Makefile2:70: recipe for target 'CMakeFiles/test_robot_bridge.dir/all' failed
make[5]: *** [CMakeFiles/test_robot_bridge.dir/all] Error 2
Makefile:130: recipe for target 'all' failed
make[4]: *** [all] Error 2
CMakeFiles/clutter_proj.dir/build.make:114: recipe for target 'clutter_proj-prefix/src/clutter_proj-stamp/clutter_proj-build' failed
make[3]: *** [clutter_proj-prefix/src/clutter_proj-stamp/clutter_proj-build] Error 2
CMakeFiles/Makefile2:107: recipe for target 'CMakeFiles/clutter_proj.dir/all' failed
make[2]: *** [CMakeFiles/clutter_proj.dir/all] Error 2
CMakeFiles/Makefile2:119: recipe for target 'CMakeFiles/clutter_proj.dir/rule' failed
make[1]: *** [CMakeFiles/clutter_proj.dir/rule] Error 2
Makefile:131: recipe for target 'clutter_proj' failed
make: *** [clutter_proj] Error 2
```
For this particular executable, it links to perception and robot_bridge, which both links against drake::drake. I have tried to make individual executable that links to either one, and it would work, but not both. I have also tried to explicitly link against libvtkCommonCore-8.0.so.1, but that ends up complaining about more vtk stuff upstream. Any ideas about what I am doing wrong here???
cc @jamiesnape @EricCousineau-TRI @stonier 
I am also blocked on this, so faster responses would be greatly appreciated.

fbudin69500, 2017-12-12:
I was trying to compile your project on my computer to reproduce your error, but it seems that it needs realsense which is not a standard package on my distribution (Ubuntu 16.04). How did you install this dependency?

siyuanfeng-tri, 2017-12-12:
right! i forgot about it. let me fix it first.

sankhesh, 2017-12-13:
@siyuanfeng-tri Could you try adding the following CMake code to [CMakeLists.txt](https://github.com/siyuanfeng-tri/rand_obj_picking/blob/f0e0e4153034391ca373bb2c30ec7632670c8dea/src/CMakeLists.txt) and see if that fixes your issue?

```cmake
find_package(VTK REQUIRED)
include(${VTK_USE_FILE}) 
```  

and the following change:

```diff
add_executable(test_robot_bridge test_robot_bridge.cc)
- target_link_libraries(test_robot_bridge perception grasp_gen robot_bridge drake::drake)
+ target_link_libraries(test_robot_bridge perception grasp_gen robot_bridge drake::drake ${VTK_LIBRARIES})
```

siyuanfeng-tri, 2017-12-13:
@fbudin69500 I just added the librealsense dependency as an external project, it should grab it and build. 
@sankhesh I added these lines to https://github.com/siyuanfeng-tri/rand_obj_picking/blob/master/src/CMakeLists.txt#L9 and uncommenting https://github.com/siyuanfeng-tri/rand_obj_picking/blob/master/src/test_robot_bridge.cc#L41 and 
https://github.com/siyuanfeng-tri/rand_obj_picking/blob/master/src/test_robot_bridge.cc#L4
gives the same linking error.

siyuanfeng-tri, 2017-12-13:
update, @sankhesh adding boost_system seems to work, but I am not quite sure that I understand what's going on here anymore... Why do I need to explicitly do vtk and boost_system? aren't them taken care of by the dependencies?

sankhesh, 2017-12-13:
@siyuanfeng-tri Awesome!

I'm not that familiar with drake package management and dependency resolution. Perhaps @fbudin69500 and @jamiesnape would have a better idea.

siyuanfeng-tri, 2017-12-13:
hey guys, i have all my stuff compiled and running on the robot now. 
but I'd really like some expert to sanity check my setup. i don't quite understand why some of the dependencies are not quite "transitive" like https://github.com/siyuanfeng-tri/rand_obj_picking/blob/master/src/CMakeLists.txt#L34

EricCousineau-TRI, 2019-02-07:
Closing, as Anzu (the spiritual success to this repo) is now Bazel-ified.

closed 2019-02-07 by EricCousineau-TRI