RobotLocomotion/drake-external-examples #74: Avoid handing out broken builds to users

jwnimmer-tri, 2017-12-20:
The https://github.com/RobotLocomotion/drake-shambhala/blob/master/drake_cmake_installed/README.md instructions point users to https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-xenial.tar.gz.  However, it appears that even if the Nightly CI fails, the package is still uploaded and the "latest" link changed to use it?  That means users will see broken nightlies, which seems awful.

Is it easy to change the build to not bump the "latest", or to have a different "latest-passing" or something?  Or worst case for now just don't upload anything?

I just had a user try to use the broken nightly and have to ask for help.

jwnimmer-tri, 2017-12-21:
Ah, I misunderstood the sequence of events.  Drake's CI publishes nightlies, and Shambhala tests against them.  So it's not a trivial fix.  I wonder if Shambhala can take over the `latest` bumping, or something.

(I guess really, we should have Shambhala in pre-merge anyway, which makes many things better.)

jwnimmer-tri, 2017-12-21:
Or I guess the real problem is that Shambhala master assumes that its setup script is compatible with Drake nightlies, which means #61 is the real problem.  I guess the Drake nightly should publish an appropriately-compatible setup script itself.

jamiesnape, 2017-12-21:
> I guess the Drake nightly should publish an appropriately-compatible setup script itself.

It is on my TODO list.

Until we have API stability and acceptance tests, we are not going to be able to guarantee that each successive night’s packages are compatible with each other. That is not unusual with nightly builds, though. The trouble is that we do not have releases yet.

jwnimmer-tri, 2017-12-21:
Those are good points.

jwnimmer-tri, 2020-09-23:
I think there's nothing left to do here.  We have releases, we publish setup scripts, and #61 tracks using the setup scripts for this CI.

closed 2020-09-23 by jwnimmer-tri