RobotLocomotion/drake-external-examples #78: runtime error on mac: can't find ignition

RussTedrake, 2018-01-10:
testing with the `drake_cmake_installed` workflow on mac.  it *almost* works.  is this something simple?

```
russt$ src/simple_continuous_time_system/simple_continuous_time_system
dyld: Symbol not found: __ZN8ignition4math5Angle4ZeroE
  Referenced from: /Users/russt/drake-build/install/lib/libdrake.so
  Expected in: /Users/russt/drake-build/install/lib/liblcm.so
 in /Users/russt/drake-build/install/lib/libdrake.so
Abort trap: 6
```

jwnimmer-tri, 2018-01-10:
RobotLocomotion/drake#7622 was recent, so possibly that disturbed something.  According to that PR, it should be trying to link / load / use `libignition_math.so` to provide that symbol, I believe.

I don't think we have Shambhala CI for macOS though, and I'm not sure how many people have tried it recently (consuming Drake as a dependency via CMake on macOS).

RussTedrake, 2018-01-10:
confirming that the `libignition_math.so` exists, but it is not listed as a dependency of `libdrake.so`

```
russs-air:lib russt$ otool -L libdrake.so
libdrake.so:
	@rpath/libdrake.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/protobuf/lib/libprotobuf.15.dylib (compatibility version 16.0.0, current version 16.1.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/local/opt/tinyxml2/lib/libtinyxml2.6.dylib (compatibility version 6.0.0, current version 6.0.0)
	/usr/local/opt/glib/lib/libglib-2.0.0.dylib (compatibility version 5401.0.0, current version 5401.3.0)
	/usr/local/opt/gettext/lib/libintl.8.dylib (compatibility version 10.0.0, current version 10.5.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1450.15.0)
	/usr/local/opt/pcre/lib/libpcre.1.dylib (compatibility version 4.0.0, current version 4.9.0)
	/usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
	/System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 158.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1450.15.0)
	/usr/local/opt/dreal/lib/libdreal.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/ibex/lib/libibex.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/clp/lib/libClpSolver.1.dylib (compatibility version 15.0.0, current version 15.11.0)
	/usr/local/opt/clp/lib/libClp.1.dylib (compatibility version 15.0.0, current version 15.11.0)
	/usr/local/opt/coinutils/lib/libCoinUtils.3.dylib (compatibility version 14.0.0, current version 14.14.0)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.5)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.0.0)
	/System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
	/usr/local/opt/ipopt/lib/libipopt.1.dylib (compatibility version 12.0.0, current version 12.8.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/nlopt/lib/libnlopt_cxx.0.dylib (compatibility version 9.0.0, current version 9.2.0)
	/usr/local/opt/vtk@8.0/lib/libvtkFiltersSources-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkFiltersGeneral-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonComputationalGeometry-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkIOGeometry-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkIOLegacy-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkIOCore-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/lz4/lib/liblz4.1.dylib (compatibility version 1.0.0, current version 1.8.0)
	/usr/local/opt/vtk@8.0/lib/libvtkIOImage-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkDICOMParser-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkmetaio-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/libpng/lib/libpng16.16.dylib (compatibility version 51.0.0, current version 51.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkRenderingOpenGL2-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkRenderingCore-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonColor-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkFiltersCore-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkFiltersGeometry-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonExecutionModel-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonDataModel-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonSystem-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonTransforms-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonMisc-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonMath-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
[... cut ...]
libBulletCollision.so			libignition_math.so			liboctomath.so
libLinearMath.so			libignition_rndf.a			libscsdir.so
libPythonQt.dylib			libignition_rndf.so			libsdformat.a
libccd.a				liblcm.so				libsdformat.so
libccd.so				liblcmtypes_bot2_core_c.a		libtinyobjloader.a
libddApp.dylib				liblcmtypes_bot2_core_c.so		libtinyobjloader.so
libdrake.so				liblcmtypes_robotlocomotion_c.a		libyaml_cpp.a
libfcl.a				liblcmtypes_robotlocomotion_c.so	libyaml_cpp.so
libfcl.so				liboctomap.a				python2.7
libfmt.a				liboctomap.so
russs-air:lib russt$
```

jamiesnape, 2018-01-10:
I will look into it momentarily. I run Mac builds locally.

jamiesnape, 2018-01-10:
```
$ curl -LO https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-mac.tar.gz
$ tar -xf drake-latest-mac.tar.gz
$ otool -L drake/lib/libdrake.so
drake/lib/libdrake.so:
	@rpath/libdrake.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/libignition_math.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/libBulletCollision.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/libLinearMath.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/libscsdir.so (compatibility version 0.0.0, current version 0.0.0)
	@loader_path/liblcm.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/protobuf/lib/libprotobuf.15.dylib (compatibility version 16.0.0, current version 16.1.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/local/opt/yaml-cpp/lib/libyaml-cpp.0.5.dylib (compatibility version 0.5.0, current version 0.5.3)
	/usr/local/opt/tinyxml2/lib/libtinyxml2.6.dylib (compatibility version 6.0.0, current version 6.0.0)
	/usr/local/opt/glib/lib/libglib-2.0.0.dylib (compatibility version 5401.0.0, current version 5401.3.0)
	/usr/local/opt/gettext/lib/libintl.8.dylib (compatibility version 10.0.0, current version 10.5.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1450.16.0)
	/usr/local/opt/pcre/lib/libpcre.1.dylib (compatibility version 4.0.0, current version 4.9.0)
	/usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
	/System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 158.0.0)
	/System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1450.16.0)
	/usr/local/opt/dreal/lib/libdreal.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/ibex/lib/libibex.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/clp/lib/libClpSolver.1.dylib (compatibility version 15.0.0, current version 15.11.0)
	/usr/local/opt/clp/lib/libClp.1.dylib (compatibility version 15.0.0, current version 15.11.0)
	/usr/local/opt/coinutils/lib/libCoinUtils.3.dylib (compatibility version 14.0.0, current version 14.14.0)
	/usr/lib/libbz2.1.0.dylib (compatibility version 1.0.0, current version 1.0.5)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.0.0)
	/System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
	/usr/local/opt/ipopt/lib/libipopt.1.dylib (compatibility version 12.0.0, current version 12.8.0)
	/usr/local/opt/mumps@5.1/lib/libdmumps.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/mumps@5.1/lib/libmpiseq.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/mumps@5.1/lib/libmumps_common.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/mumps@5.1/lib/libpord.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/nlopt/lib/libnlopt_cxx.0.dylib (compatibility version 9.0.0, current version 9.2.0)
	/usr/local/opt/vtk@8.0/lib/libvtkFiltersSources-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkFiltersGeneral-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonComputationalGeometry-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkIOGeometry-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkIOLegacy-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkIOCore-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/lz4/lib/liblz4.1.dylib (compatibility version 1.0.0, current version 1.8.0)
	/usr/local/opt/vtk@8.0/lib/libvtkIOImage-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkDICOMParser-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkmetaio-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/X11/lib/libpng16.16.dylib (compatibility version 43.0.0, current version 43.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkRenderingOpenGL2-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkRenderingCore-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonColor-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkFiltersCore-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkFiltersGeometry-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonExecutionModel-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonDataModel-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonSystem-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonTransforms-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
[... cut ...]
	/usr/local/opt/vtk@8.0/lib/libvtkCommonMath-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtkCommonCore-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk@8.0/lib/libvtksys-8.0.1.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/glew/lib/libGLEW.2.1.dylib (compatibility version 2.1.0, current version 2.1.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.0)
```

Can you try the latest nightly package?

jamiesnape, 2018-01-11:
I think this has something to do with the Bazel version, but still trying to work out what changed and when.

RussTedrake, 2018-01-11:
fwiw, it looks like homebrew bumped it's bazel version last week (so I got 0.9):
```
russs-air:mac russt$ brew info bazel
bazel: stable 0.9.0 (bottled)
Google's own build tool
https://bazel.build/
/usr/local/Cellar/bazel/0.9.0 (12 files, 93.2MB) *
  Poured from bottle on 2018-01-06 at 06:24:55
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/bazel.rb
==> Requirements
Required: java = 1.8 ✔, macOS >= 10.10 ✔
==> Caveats
Bash completion has been installed to:
  /usr/local/etc/bash_completion.d

zsh completions have been installed to:
  /usr/local/share/zsh/site-functions
```

jamiesnape, 2018-01-12:
I think the problem is an unreported bug in Bazel (not sure when it was introduced yet) that will be fixed in either 0.9.1 or 0.10.0 that relates to the toolchain. Our custom toolchain has diverged (and in fact was taken from the one that Bazel uses to build itself rather than what it uses to build other projects), which I will remedy later.

RussTedrake, 2018-01-13:
I've installed bazel@0.6 (which has 0.6.1) and inserted it in my path, run `bazel clean` in drake, then `make clean; make` in `drake-build`, and `make clean; make` in `shambhala/drake_cmake_installed/build`.  But I still get the same error.
```
Russs-MacBook-Air:build russt$ ./src/simple_continuous_time_system/simple_continuous_time_system
dyld: Symbol not found: __ZN8ignition4math5Angle4ZeroE
  Referenced from: /Users/russt/drake-build/install/lib/libdrake.so
  Expected in: /Users/russt/drake-build/install/lib/liblcm.so
 in /Users/russt/drake-build/install/lib/libdrake.so
Abort trap: 6
Russs-MacBook-Air:build russt$ bazel version
Build label: 0.6.1-homebrew
Build target: bazel-out/darwin_x86_64-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Wed Jan 10 21:28:05 2018 (1515619685)
Build timestamp: 1515619685
Build timestamp as int: 1515619685
```

jamiesnape, 2018-01-15:
I can reproduce this now with an up-to-date custom toolchain, so I know exactly the breakage. Just iterating through Bazel versions to find if one is not broken.

jamiesnape, 2018-01-15:
Not good news so far looking for a stable version. About to build the `release-0.10.0` branch, which looks promising.

jamiesnape, 2018-01-15:
Until Bazel 0.10.0 is out, the workaround is:
```
brew tap robotlocomotion/director
brew install bazel@0.10
export PATH=/usr/local/opt/bazel@0.10/bin:$PATH
```

jamiesnape, 2018-02-20:
Now that 0.10.0+ is out, I think we can mark this resolved.

closed 2018-02-20 by jamiesnape