RobotLocomotion/drake-external-examples #86: Is `drake_cmake_installed/test_find_resource` tested on CI?

EricCousineau-TRI, 2018-01-19:
Do we have a guarantee that this test is run on CI?
If not, can we require it?

I ask because I had copied from `find_resource.cc.in` for `test_find_resource.py` for #85, and then seemed to have run into https://github.com/RobotLocomotion/drake/issues/7809

When I tried building / running the test on my machine, it reports `GTEST_NOT_FOUND`, even though I have `libgtest-dev` installed.

\cc @jamiesnape @fbudin69500

jamiesnape, 2018-01-19:
`libgtest-dev` only installs the source. There was a long thread on a PR or an issue about this and I think @stonier decided to punt in the end.

EricCousineau-TRI, 2018-01-19:
In that case, can we make `find_resource` not need GTest?

jamiesnape, 2018-01-19:
We could, but I it depends whether we plan on porting more tests across from drake.

PR #87 (2018-02-14): find_resource: Don't use GTest if it's not easily enabled.
Closes #86.

This is a minor change to ensure this test is run, given that `find_package(GTest)` is not functional out-of-the-box.

I can live with/without it, but would like to ensure this test is run.

\cc @jamiesnape

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake-shambhala/87)
closed 2018-02-14 by EricCousineau-TRI