RobotLocomotion/drake-external-examples #99: Linking Strategy

stonier, 2018-03-08:
Without diving into any plan yet, would like to fathom exactly what is going on. There are some aspects of the linking I haven't come across before and some of this is having surprising effects for us (in Delphyne).

Minimal Example: Particle demo at https://github.com/stonier/drake-shambhala/tree/stonier/link_paths.

The only differences here is the library is built shared and both library and particle executable are installed. Build with:

```
cmake -DCMAKE_INSTALL_PREFIX=../install -Ddrake_DIR=/opt/drake/lib/cmake/drake ..
make
make install
```

Resulting links in the build directory:

```
ldd build/src/particles/uniformly_accelerated_particle
        libparticles.so => /mnt/mervin/workspaces/shambhala/drake_cmake_installed/build/src/particles/libparticles.so (0x00007fea3504e000)
        /opt/drake/lib/libdrake.so (0x00007fea33a02000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fea33680000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fea33377000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fea33161000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fea32d97000)
        /opt/drake/lib/libdrake_bullet_collision.so (0x00007fea32a88000)
        /opt/drake/lib/libdrake_bullet_linear_math.so (0x00007fea32864000)
        /opt/drake/lib/libdrake_ignition_rndf0.so (0x00007fea3262f000)
        /opt/drake/lib/libdrake_ignition_math4.so (0x00007fea323f6000)
        /opt/drake/lib/libdrake_lcm.so (0x00007fea321dd000)
        libvtkFiltersSources-8.0.so.1 => /opt/drake/lib/libvtkFiltersSources-8.0.so.1 (0x00007fea31d08000)
```

In the install directory:

```
ldd install/bin/uniformly_accelerated_particle
        libparticles.so => not found
        libgflags.so.2 => /usr/lib/x86_64-linux-gnu/libgflags.so.2 (0x00007f9c78706000)
        /opt/drake/lib/libdrake.so (0x00007f9c772db000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9c76f59000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9c76c50000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9c76a3a000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9c76670000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9c76453000)
        libdrake_ignition_rndf0.so => /opt/drake/lib/libdrake_ignition_rndf0.so (0x00007f9c7621e000)
        libdrake_ignition_math4.so => /opt/drake/lib/libdrake_ignition_math4.so (0x00007f9c75fe5000)
        libvtkFiltersSources-8.0.so.1 => /opt/drake/lib/libvtkFiltersSources-8.0.so.1 (0x00007f9c75d2d000)
```

Questions:

* Why is the `libdrake.so` path locked in?
* Why do the `libdrake.so` and dependency paths not stripped on install? (not rpaths?)

jamiesnape, 2018-03-08:
Apparently this also a Bazel problem. Of the binaries that we distribute, only the ones build using CMake have `SONAME` set. We can fix that too.

stonier, 2018-03-10:
Tested on the preceding PR which should have solved the irregularities.

closed 2018-03-10 by stonier