RobotLocomotion/drake-external-examples #107: Install drake binary from the source

junhyeokahn, 2018-06-14:
Hello,
I am using MAC OS X, and want to use drake_cmake_installed.
In the README file, there is an instruction to install drake binaries for UBUNTU but not for MAC.
I would like to install binaries from the source, so want to use the method similar to ```sudo make install``` in ```cmake``` but I don't know the equivalent method  in ```bazel```.

Is there any way for me to install binaries from the drake source by using a method similar to ```make install``` in MAC?

jwnimmer-tri, 2018-06-14:
Disclaimer: I haven't personally tried any of the below, but maybe it helps...

See also http://drake.mit.edu/from_binary.html.  In the https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_cmake_installed#instructions if you replace `-xenial` in the targz filename with `-mac` you would get the pre-compiled binary release for macOS.  If I recall correctly, the https://github.com/RobotLocomotion/drake/tree/master/setup/mac/binary_distribution prereqs script is included in the targz, but if not you can refer to that source copy instead.

I'm not sure what else has to change vs Ubuntu, but maybe that's a start.

RussTedrake, 2018-06-14:
fwiw, the installation instructions here: http://underactuated.mit.edu/underactuated.html?chapter=drake where used successfully for the last ~6 mos (and contain mac)

junhyeokahn, 2018-06-14:
@RussTedrake @jwnimmer-tri  Thanks, if I want to use gurobi, mosek, snopt, should I add extra snippets on installation?

junhyeokahn, 2018-06-14:
Thank you so much!

closed 2018-06-14 by junhyeokahn