RobotLocomotion/drake-external-examples #108: drake_bazel_external vs commercial solvers

jwnimmer-tri, 2018-06-14:
The drake_bazel_external documentation doesn't give any hint about how to turn on the commercial solvers for mathematical programs (SNOPT, MOSEK, Gurobi) when using Drake as a Bazel external.  We should document it.

jwnimmer-tri, 2018-06-14:
The easy answer to recommend is that in the downstream `tools/bazel.rc` file within their project, to add something like this:

```
# Enable Gurobi by default.
build --define=WITH_GUROBI=ON
```

Maybe just some commented-out lines in our example `drake_bazel_external/tools/bazel.rc` would be good enough.

pvarin, 2018-06-16:
It might also make sense to mention that we need to add the license file environment variable to allow tests to work
```
test --test_env=GRB_LICENSE_FILE
```

jamiesnape, 2018-06-18:
> The drake_bazel_external documentation doesn't give any hint about how to turn on the commercial solvers for mathematical programs (SNOPT, MOSEK, Gurobi) when using Drake as a Bazel external. We should document it.

Indeed. It was, of course, a deliberate omission originally, though.

junhyeokahn, 2018-07-15:
Hello, I have additional question.
I would like to use ```mosek``` or ```gurobi``` with ```drake_cmake_installed```.
I have a mosek license installed in my computer.
But the function ```drake::solvers::MosekSolver::AcquireLicense()``` returns ```nullptr```.
Is there any way to specify mosek license with ```drake_cmake_installed```?

tyler-yankee, 2025-05-02:
I considered the idea of providing similar documentation in `drake_cmake_external` for the corresponding CMake options, but we already provide something akin to this for the open-source solvers:
https://github.com/RobotLocomotion/drake-external-examples/blob/52abaa1d66e3db6d8bd04ad9b4e1c11da9c6d5e9/drake_cmake_external/CMakeLists.txt#L123-L126

So probably it's fine as-is?

jwnimmer-tri, 2025-05-02:
Sure.  The less time we spend on this issue, the better.  I was planning on closing it as "already fixed" by the docs we recently added at https://drake.mit.edu/from_source.html#building-with-bazel, but you self-added onto the board without me getting to it first.

PR #391 (2025-06-05): [drake_bazel_external] Add documentation for proprietary solvers
Closes #108.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake-external-examples/391)
closed 2025-06-05 by tyler-yankee