RobotLocomotion/drake-external-examples #111: Travis CI: Failling due to `libprotobuf.15.dylib`

EricCousineau-TRI, 2018-06-21:
It seems that protobuf is causing TravisCI to fail:
One example from this build: https://travis-ci.org/RobotLocomotion/drake-shambhala/jobs/394709115#L3073-L3078
```
3: Test command: /Users/travis/build/RobotLocomotion/drake-shambhala/drake_cmake_installed/build/src/particles/particle_test
3: Test timeout computed to be: 1500
3: dyld: Library not loaded: /usr/local/opt/protobuf/lib/libprotobuf.15.dylib
3:   Referenced from: /opt/drake/lib/libdrake.so
3:   Reason: image not found
```

Should we be pinning our `protobuf` version?

jamiesnape, 2018-06-21:
Protobuf got updated sometime between when the nightly packages were built and when the nightly shambhala job ran. Should be resolved when the next nightly packages are built. Not much we can do about those situations.

EricCousineau-TRI, 2018-06-22:
Confirmed that TravisCI is now passing.

closed 2018-06-22 by EricCousineau-TRI