RobotLocomotion/drake-external-examples #144: Problem with output_port_value.h

atabakd, 2019-10-10:
I am trying to build `drake_cmake_installed` but run into this issue:

Scanning dependencies of target find_resource_example
[  5%] Building CXX object src/find_resource/CMakeFiles/find_resource_example.dir/find_resource_example.cc.o
[ 10%] Linking CXX executable find_resource_example
[ 10%] Built target find_resource_example
Scanning dependencies of target particles
[ 15%] Building CXX object src/particles/CMakeFiles/particles.dir/particle.cc.o
[ 21%] Building CXX object src/particles/CMakeFiles/particles.dir/utilities.cc.o
[ 26%] Linking CXX static library libparticles.a
[ 26%] Built target particles
Scanning dependencies of target uniformly_accelerated_particle
[ 31%] Building CXX object src/particles/CMakeFiles/uniformly_accelerated_particle.dir/uniformly_accelerated_particle.cc.o
[ 36%] Linking CXX executable uniformly_accelerated_particle
[ 36%] Built target uniformly_accelerated_particle
Scanning dependencies of target particle_test
[ 42%] Building CXX object src/particles/CMakeFiles/particle_test.dir/particle_test.cc.o
/home/user/GitHub/drake-external-examples/drake_cmake_installed/src/particles/particle_test.cc:39:10: fatal error: drake/systems/framework/output_port_value.h: No such file or directory
 #include <drake/systems/framework/output_port_value.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
src/particles/CMakeFiles/particle_test.dir/build.make:62: recipe for target 'src/particles/CMakeFiles/particle_test.dir/particle_test.cc.o' failed
make[2]: *** [src/particles/CMakeFiles/particle_test.dir/particle_test.cc.o] Error 1
CMakeFiles/Makefile2:1146: recipe for target 'src/particles/CMakeFiles/particle_test.dir/all' failed
make[1]: *** [src/particles/CMakeFiles/particle_test.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

Maybe now that drake has stable releases, there's no need to pull the nightly versions and here can have a simple release too.

jwnimmer-tri, 2019-10-10:
Apologies -- it looks like our CI system lost track of googletest, and thus soft-failed back to ignoring some code, and thus the code bitrotted.  We'll need to fix CI and then update this example to use the current include paths and API spellings.

jamiesnape, 2019-10-10:
Easiest thing is to drop the requirement for GTest per
https://github.com/RobotLocomotion/drake-external-examples/blob/87c9fad0e166e8480c4fa00880ab610ce4026df5/drake_cmake_installed/src/particles/CMakeLists.txt#L48

atabakd, 2019-10-10:
Thanks for the quick reply.
Ignoring particle example in `drake-external-examples/drake_cmake_installed/src/CMakeLists.txt` allowed me to build though.
```
# -*- mode: cmake -*-
# vi: set ft=cmake :

# Copyright (c) 2017, Massachusetts Institute of Technology.
# All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#
# * Redistributions of source code must retain the above copyright notice, this
#   list of conditions and the following disclaimer.
#
# * Redistributions in binary form must reproduce the above copyright notice,
#   this list of conditions and the following disclaimer in the documentation
#   and/or other materials provided with the distribution.
#
# * Neither the name of the copyright holder nor the names of its contributors
#   may be used to endorse or promote products derived from this software
#   without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
# AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
# IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
# ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
# LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
# CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
# SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
# INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
# CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
# ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
# POSSIBILITY OF SUCH DAMAGE.

add_subdirectory(find_resource)
#add_subdirectory(particles)
add_subdirectory(pcl)
add_subdirectory(simple_bindings)
add_subdirectory(simple_continuous_time_system)
```

PR #145 (2019-10-21): Vendor googletest and fix failing tests
I still would like to remove the dependency on googletest, but this is the easiest way to get things building again for now.

Fixes #144.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake-external-examples/145)
closed 2019-10-10 by atabakd