RobotLocomotion/drake-external-examples #153: request: drake_bazel_installed should pull the install directory from an env variable

RussTedrake, 2020-01-10:
to be consistent with 
https://github.com/RobotLocomotion/drake-external-examples/blob/70040a3a88b3b320823e5353f7cc759602391d86/drake_bazel_external/WORKSPACE#L52

(and because it's the workflow i plan to use in my textbook)

jwnimmer-tri, 2020-09-23:
@RussTedrake I just noticed that this issue got overlooked the build system issue triage.  (I forgot to look within this repository.)  Can you help clarify whether this is still relevant (I assume "yes"), and what urgency / timeline would be appropriate?

RussTedrake, 2020-09-24:
Yes, still relevant.  No urgency.  I've added support for this in my own repos, and @jamiesnape and I thought we should propagate it back upstream, in the spirit of not having multiple versions and "if it's not working for the use cases we know about, it's probably not working".  

https://github.com/RussTedrake/manipulation/blob/master/WORKSPACE.bazel#L63

jwnimmer-tri, 2022-09-29:
For the record, `drake_bazel_external` already has the approximate logic for this, we just need to copy it over here.  Users in the meantime who need this feature can peek there for a hint.

jwnimmer-tri, 2024-03-06:
Per #286, closing this as "won't fix".  Our canonical bzlmod workflow is to use `http_archive`, which can accept `file://` URLs if necessary.  Also bzlmod has `local_path_override` to override to use a local file, instead.  To the extent that users want to control that with environment variables for some local IDE sugar, that's out of scope for this repository.

closed 2024-03-06 by jwnimmer-tri