RobotLocomotion/drake-external-examples #154: drake_bazel_installed workflow fails on importing with `cpp_template.py`

RussTedrake, 2020-01-10:
Just looked into this w/ @jwnimmer-tri on my desktop, and reduced it ot the following minimal reproduction...

Running the code below from bazel results in a huge spew ending with
```
 what():  SystemError: <built-in method __contains__ of dict object at 0x7fe56e9a4438> returned a result with an error set
```

code:
```
import trace

tracer = trace.Trace(trace=1, count=0, ignoredirs=[])
tracer.run('from pydrake.multibody.plant import AddMultibodyPlantSceneGraph')
```

EricCousineau-TRI, 2020-01-10:
Hm... IIRC, this may be due to some Python package leaking in, but it occurred when I was using Anzu (don't remember repro'ing in Drake, but it probably could have).

In my case, it was `pytest` getting installed by `pip install --user` or `sudo pip install` (which shouldn't be done in general).

Unfortunately, `pybind11` can swallow errors at import, so it makes it a bit painful to try and debug... I believe @thduynguyen ran into this same problem on his machine (Dec. 20th), but I forgot what it was he installed.

Can you check your `numpy` version?
My Ubuntu-installed version:
```
$ python3 -c 'import numpy as m; print(m.__version__); print(m.__file__)'
1.13.3
/usr/lib/python3/dist-packages/numpy/__init__.py
```

RussTedrake, 2020-01-11:
I got the same error on my home machine, and confirmed my numpy version matches yours
```
russt@Puget-179850-01:~/manipulation/src/planar_iiwa$ python3 -c 'import numpy as m; print(m.__version__); print(m.__file__)'
1.13.3
/usr/lib/python3/dist-packages/numpy/__init__.py
```

RussTedrake, 2020-01-11:
I should also note that if i run the same scripts by putting the installed drake in my PYTHONPATH and call them manually via python3 in the shell (as opposed to through bazel), then they run fine.

RussTedrake, 2020-01-12:
Also, it fails on my mac, with
```
Target //src/planar_iiwa:viz_test up-to-date:
  bazel-bin/src/planar_iiwa/viz_test
INFO: Elapsed time: 25.919s, Critical Path: 0.51s
INFO: 0 processes.
INFO: Build completed successfully, 5 total actions
INFO: Build completed successfully, 5 total actions
exec ${PAGER:-/usr/bin/less} "$0" || exit 1
Executing tests from //src/planar_iiwa:viz_test
-----------------------------------------------------------------------------
libc++abi.dylib: terminating with uncaught exception of type pybind11::error_already_set: SystemError: <built-in method __contains__ of dict object at 0x10cfc48c0> returned a result with an error set
Abort trap: 6
```

EricCousineau-TRI, 2020-01-13:
Gotcha. I can repro locally on my machine with:
```
$ git clone git@github.com:RussTedrake/manipulation.git -b bazel  # 4560088
$ cd manipulation
$ bazel run //src/planar_iiwa:viz_test
```

Looking at it briefly.

EricCousineau-TRI, 2020-01-13:
Testing this out, but using a merge with latest pybind (using https://github.com/RobotLocomotion/drake/pull/12572), gives a more useful error message:
```
Traceback (most recent call last):
  File ".../drake_external_examples/apps/simple_logging_example.py", line 36, in <module>
    import pydrake.multibody.plant
ImportError: ImportError: ImportError: AttributeError: module 'drake' has no attribute 'lcmt_contact_results_for_viz'
```

EricCousineau-TRI, 2020-01-13:
Looking at drake-generated `repo.bzl`, the Bazel legacy auto-init module for `@drake` is shadowing the "true" drake LCM types package located in `@drake//:.lib/python3.6/site-packages/drake`:
```
import drake
print(drake.__file__)  # Shows {target.runfiles}/drake/__init__.py
```

Workaround is to just move those symlinks to the repository spot.

EricCousineau-TRI, 2020-01-15:
Resolved by https://github.com/RobotLocomotion/drake/pull/12573

closed 2020-01-15 by EricCousineau-TRI