RobotLocomotion/drake-external-examples #159: Need to fail-fast on deprecations

EricCousineau-TRI, 2020-01-30:
Deprecations generally slip through, and cause breakage when their removed.

jwnimmer-tri, 2020-02-10:
I broke drake-external-examples master again because of this via https://reviewable.io/reviews/RobotLocomotion/drake/12680. Drake removed a deprecated text_logging_gflags header, but it remained used by drake-external-examples until it was fully removed and thus broke the build.

EricCousineau-TRI, 2020-03-19:
Per f2f: C++ should be easy (`-Wdeprecations` or what not), if we're not already doing that.

For Python, will be annoying (env vars with test runners, updating each testing mode, or parsing stdout), but doable.

jwnimmer-tri, 2024-09-05:
I think when I work on https://github.com/RobotLocomotion/drake/issues/19167 soon, I should circle back and toggle it here as well.