RobotLocomotion/drake-external-examples #161: Not clear how to point drake_bazel_external at a release

ggould-tri, 2020-02-13:
The `drake_bazel_external/WORKSPACE` file indicates how to pin to a specific drake commit, or to live on master, but not how to pin to a numbered release.  The naive approach of putting the release tag name into `DRAKE_COMMIT` does not work.

jwnimmer-tri, 2020-02-13:
I'd guess that `COMMIT = "v0.15.0"` would download successfully -- the https://github.com/RobotLocomotion/drake/releases/tag/v0.15.0 links to https://github.com/RobotLocomotion/drake/archive/v0.15.0.tar.gz as its canonical answerl.

However, when using a v-numbered release, the `strip_prefix` argument needs to lose the leading `v` since the to-level folder name in the source archive uses the bare version number, without the `v`.

ggould-tri, 2020-02-13:
Yes, I believe that the strip_prefix bit is the problem.  Should be an easy enough fix, I just scribed the bug as I was working through it -- a naive Drake user will need a bit of guidance.

commit f5518310 (2020-02-13): Allow use of "vN.NN.N" releases in DRAKE_COMMIT

Fixes #161
commit 0dbb5551 (2020-02-13): Allow use of "vN.NN.N" releases in DRAKE_COMMIT (#162)

Fixes #161
closed 2020-02-13 by jamiesnape