RobotLocomotion/drake-external-examples #185: drake_bazel_external fails on mac

soonho-tri, 2020-09-11:
Repro Step
-----------

```bash
cd drake_bazel_external
bazel test //... --c dbg
```

Output
-------

```
INFO: From Testing //apps:simple_adder_test:
==================== Test output for //apps:simple_adder_test:
Output values: 110 110
Assertion failed: (aLhs.rows() == aRhs.rows() && aLhs.cols() == aRhs.cols()), function CwiseBinaryOp, file external/eigen/include/_usr_local_Cellar_eigen_3.3.7_include_eigen3/Eigen/src/Core/CwiseBinaryOp.h, line 110.
================================================================================
```

My Env
-------

Output of `brew config`
```
HOMEBREW_VERSION: 2.5.0-84-g9e697bd
ORIGIN: https://github.com/Homebrew/brew
HEAD: 9e697bd8ad5f3f66897a86164d3cdd1d8db681f9
Last commit: 5 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: edfeb931c1c572179ab7cecb8fbdfb08c133dc8c
Core tap last commit: 73 minutes ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_EDITOR: emacsclient
HOMEBREW_MAKE_JOBS: 24
Homebrew Ruby: 2.6.3 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: 24-core 64-bit ivybridge
Clang: 11.0 build 1103
Git: 2.28.0 => /usr/local/bin/git
Curl: 7.64.1 => /usr/bin/curl
Java: 1.8.0_112
macOS: 10.15.6-x86_64
CLT: 1103.0.32.62
Xcode: 11.6
XQuartz: 2.7.11 => /opt/X11
```

It's using `eigen: stable 3.3.7 (bottled), HEAD`.

jwnimmer-tri, 2020-09-11:
Do you happen to know if the same occurs on Ubuntu also?

soonho-tri, 2020-09-11:
In CI, for mac, we are only checking `drake_cmake_installed` (https://github.com/RobotLocomotion/drake-external-examples/blob/master/scripts/continuous_integration/github_actions/macos_catalina/build_test#L34). I guess it's worth checking the others as well (e.g. `drake_bazel_external`).

soonho-tri, 2020-09-11:
> Do you happen to know if the same occurs on Ubuntu also?

Yes.

```c++
INFO: From Testing //apps:simple_adder_test:
==================== Test output for //apps:simple_adder_test:
Output values: 110 110
simple_adder_test: external/eigen/include/_usr_include_eigen3/Eigen/src/Core/CwiseBinaryOp.h:110: Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::CwiseBinaryOp(const Lhs&, const Rhs&, const BinaryOp&) [with BinaryOp = Eigen::internal::scalar_difference_op<double, double>; LhsType = const Eigen::Block<const Eigen::Matrix<double, -1, -1>, -1, -1, true>; RhsType = const Eigen::Matrix<double, -1, 1>; Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::Lhs = Eigen::Block<const Eigen::Matrix<double, -1, -1>, -1, -1, true>; Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::Rhs = Eigen::Matrix<double, -1, 1>]: Assertion `aLhs.rows() == aRhs.rows() && aLhs.cols() == aRhs.cols()' failed.
================================================================================
Target //apps:simple_adder_test up-to-date:
  bazel-bin/apps/simple_adder_test
```

jwnimmer-tri, 2020-09-11:
> In CI, for mac, we are only checking drake_cmake_installed ...

I believe that's because from-source builds are too slow for the CI here.

closed 2020-09-11 by EricCousineau-TRI

jamiesnape, 2020-09-11:
No reason we cannot run some Ubuntu debug versions of `*_external` on CI here.

PR #186 (2020-09-11): Update simple_adder_test.cc
Close #185

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake-external-examples/186)
commit 8a53b4c8 (2020-09-11): Update simple_adder_test.cc (#186)

Close #185