RobotLocomotion/drake-external-examples #209: [drake_bazel_installed] Compile Error when using a stable Drake installed from APT (focal:0.35.0-1)

pvmilk, 2021-11-04:
I compiled **drake_bazel_installed** by setting INSTALLED_DRAKE_DIR="/opt/drake/" in **drake_bazel_installed/WORKSPACE**.

Drake was installed from apt-get install in Ubuntu-focal. The version is as follows:
```https://drake-apt.csail.mit.edu/focal focal/main amd64 drake-dev amd64 0.35.0-1 ```

Compile with the following command... and got the error below. 
```
$ cd $ROOT/drake_bazel_installed
$ bazel build --jobs=3 //apps:all

Starting local Bazel server and connecting to it...
INFO: Analyzed 4 targets (30 packages loaded, 1890 targets configured).
INFO: Found 4 targets...
ERROR: /home/USER/ide-workspace/drake-external-examples/drake_bazel_installed/apps/BUILD.bazel:46:8: Middleman _middlemen/apps_Sfind_Uresource_Utest-runfiles failed: missing input file 'external/drake/examples/quadrotor/LICENSE', owner: '@drake//examples:quadrotor/LICENSE'
ERROR: /home/USER/ide-workspace/drake-external-examples/drake_bazel_installed/apps/BUILD.bazel:46:8 Middleman _middlemen/apps_Sfind_Uresource_Utest-runfiles failed: 10 input file(s) do not exist
INFO: Elapsed time: 7.751s, Critical Path: 0.15s
INFO: 11 processes: 11 internal.
FAILED: Build did NOT complete successfully
```

I tracked it down and found out the error is more to do with how the *.deb package was packages.
I believed that some LICENSE files have been removed during stable release of the *.deb package, but the file **/opt/drake/share/drake/repo.bzl** has not been updated accordingly; hence the error. 

P.S.
* I tried to compile by letting bazel install the nightly build (without setting INSTALLED_DRAKE_DIR). It works fine. 
* Sorry, if the issue should be reported in the main repository.

jwnimmer-tri, 2021-11-04:
Thanks for the report, it's fine to be here.  (I'll move it to a Drake issue later if need be.)

When I wrote the `drake_bazel_installed` example, I only imagined that it would be used versus the binary release tarballs from https://drake.mit.edu/from_binary.html, not apt.  If you want to try those for now, I think they would work.

However, the idea of using it vs apt releases is a good one.  I'll see if we can make that work.

jwnimmer-tri, 2024-03-06:
Per #286, the `repo.bzl` file is no longer used, so I don't think it can cause problems anymore.

closed 2024-03-06 by jwnimmer-tri