RobotLocomotion/drake-external-examples #210: [drake_bazel_installed] C++ Examples for Bazel when Using with Installed Package

pvmilk, 2021-11-10:
Are there any example of how to use bazel when install drake as the external library?

I checked out **/opt/drake/share/drake/repo.bzl** and found out the following:
```
147 # TODO(jwnimmer-tri) Use the correct cc_import phrasing, then mark this public.
148 filegroup(
149     name = ".drake_shared_library",
150     data = glob([
151     ┆   ".lib/*.so",
152     ┆   ".lib/*.so.*",
153     ]),
154 )
```

I am pretty new to bazel, but I think this meant that I have to write a bazel rule by myself. 
Do you have an example of such bazel rule which would allow me to do the following?
```
    deps = [
          "@drake//:drake_shared_library",
    ],
```

P.S. I am aware of "drake_bazel_external", but compiling drake simply took too long on my machine.

jwnimmer-tri, 2021-11-10:
That's a missing feature; tracked as https://github.com/RobotLocomotion/drake/issues/12749.  Apologies, but I don't think I have an example of fixing this yet.  I'll see if we can fix the TODO before the next release, in any case.

I am a :heart_eyes: of Bazel so I hate to say this, but for the moment as a workaround note that https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed or https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed_apt are available and allow for using pre-compiled builds of Drake (either stable tags or nightlies).

pvmilk, 2021-11-10:
Thank you for a reply & response. Glad to see that others also encounter similar issue. Please take your time. 

I am pretty new to Bazel and decided stick to it for the current project.
I am also trying to include other external libraries in the same way and just encounter the issues of the third-parties dependencies. Hope that won't be the issue in drake!

PR #226 (2022-05-11): Add find_resource_test C++ example
Closes #210
Context: https://github.com/RobotLocomotion/drake-ros/pull/75#pullrequestreview-955704403

+@EricCousineau-TRI for review please

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake-external-examples/226)
closed 2022-05-11 by EricCousineau-TRI