RobotLocomotion/drake-external-examples #214: [drake-bazel-installed] Python examples all fail to import pydrake

mposa, 2022-01-15:
Simple, but serious breaking error. I tried to use this workflow today and it did not work at all.

All of the examples in `drake_bazel_installed/apps` fail to import pydrake when run.

closed 2022-01-15 by mposa

jwnimmer-tri, 2022-01-18:
I'm curious to see if I can improve the documentation or error messages in this case.  I went to try to find the place where Ubuntu 18 was lurking, but didn't immediately see it.  Do you remember what part of the code the confusion came from?

mposa, 2022-01-18:
In the WORKSPACE file:

https://github.com/RobotLocomotion/drake-external-examples/blob/be73cfe7567fcaa8ca103f4c0b7810d9076a430c/drake_bazel_installed/WORKSPACE#L62

jwnimmer-tri, 2022-01-19:
Aha!  That is quite hidden; I missed it.  I'll open a PR to make it more clear.