RobotLocomotion/drake-external-examples #216: [drake_cmake_installed] CMake error when building in release mode

pangtao22, 2022-02-14:
I've been structuring my research projects using the `drake_cmake_installed` pattern. About 1 and a half years ago, I started to get this error when I was running cmake in release mode:

```
amazon@amazon-system-product-name:~/ClionProjects/drake-external-examples/drake_cmake_installed/build$ cmake .. -DCMAKE_PREFIX_PATH=~/drake-build/install -DCMAKE_BUILD_TYPE=release
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3.6") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable version "3.8.10", minimum required is "3.6") 
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
CMake Error at /home/amazon/drake-build/install/lib/cmake/pybind11/pybind11Tools.cmake:204 (pybind11_strip):
  Unknown CMake command "pybind11_strip".
Call Stack (most recent call first):
  src/simple_bindings/CMakeLists.txt:35 (pybind11_add_module)

-- Configuring incomplete, errors occurred!
See also "/home/amazon/ClionProjects/drake-external-examples/drake_cmake_installed/build/CMakeFiles/CMakeOutput.log".
See also "/home/amazon/ClionProjects/drake-external-examples/drake_cmake_installed/build/CMakeFiles/CMakeError.log".
```

It seems that `pybind11_strip` is a part of the pybind cmake files, but it's not installed in the pybind bundled with drake? Our workaround has been simply commenting out the line in cmake that calls `pybind11_strip`... which is not ideal, and could be the reason why we're getting "weak symbol" warnings after the bindings are built.

pangtao22, 2022-11-28:
Adding [these lines](https://github.com/pybind/pybind11/blob/9c0aa69937dbb9d7023123d5d98914855432982c/tools/pybind11Common.cmake#L383-L397) from pybind11 to `drake/tools/workspace/pybind11/pybind11-config.cmake` (e.g. [like this](https://github.com/pangtao22/drake/blob/ba6ba4f5f7c3d22d18778519ce48893a81a7409f/tools/workspace/pybind11/pybind11-config.cmake#L113-L125)) can solve the problem, but I'm not sure if that's the correct way to do it.

EricCousineau-TRI, 2022-11-28:
Howdy Pang! I am unable to reproduce your error. On my system, using v1.10.0:
```sh
$ cmake -DCMAKE_PREFIX_PATH=~/.local/opt/drake/v1.10.0 ..
# No problems.
$ make -j
# No problems.
```

I believe this may be a system-level dependency issue (either in `/usr` or `/usr/local`).

Is there any way you could reproduce this in a Docker or [Apptainer](https://apptainer.org/) container?
(here is how I use Apptainer, for better or worse: [repro/.../apptainer_stuff](https://github.com/EricCousineau-TRI/repro/tree/2cff386a1c0dd6e19c0fe3de4d314a62f7dcbe76/shell/apptainer_stuff))

pangtao22, 2022-11-29:
I'm able to reproduce the error in a docker container:
https://github.com/pangtao22/quasistatic_simulator/actions/runs/3570809861/jobs/6002118367 

The cause is the `-DCMAKE_BUILD_TYPE=Release` flag in the cmake command, which is missing in your run @EricCousineau-TRI . 

Thanks for looking into this!

closed 2024-09-05 by jwnimmer-tri