RobotLocomotion/drake-external-examples #231: [drake_cmake_installed] CMake error when make in release mode

Yingfan99327, 2022-06-17:
When I try to run make after I run cmake -DCMAKE_PREFIX_PATH=/opt/drake ../drake_cmake_installed I met some problems as follows:
/usr/bin/ld: warning: libgfortran.so.5, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libspdlog.so.1, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libamd.so.2, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libnlopt_cxx.so.0, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libipopt.so.1, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libdmumps_seq-5.2.1.so, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libClpSolver.so.1, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libClp.so.1, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libCoinUtils.so.3, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libGLEW.so.2.1, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libdouble-conversion.so.3, needed by /opt/drake/lib/libdrake.so, not found (try using -rpath or -rpath-link)
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_set_ftol_rel'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewDeleteQueries'
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_set_ftol_abs'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewBindBuffer'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewGenRenderbuffers'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewBindBufferBase'
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_set_maxtime'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_concat_string@GFORTRAN_8'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewEndTransformFeedback'
//opt/drake/lib/libdrake_ibex.so: undefined reference to `ClpSimplex::dual(int, int)'
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_set_xtol_abs1'
/opt/drake/lib/libdrake.so: undefined reference to `std::filesystem::__cxx11::directory_iterator::operator*() const@GLIBCXX_3.4.26'
/opt/drake/lib/libdrake.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
/opt/drake/lib/libdrake.so: undefined reference to `ClpModel::setObjectiveOffset(double)'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewGetNamedRenderbufferParameteriv'
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_add_inequality_constraint'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__GLEW_VERSION_3_2'
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_destroy'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewBufferData'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_st_read@GFORTRAN_8'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_st_close@GFORTRAN_8'
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_set_upper_bounds'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_st_rewind@GFORTRAN_8'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewUniform3fv'
/opt/drake/lib/libdrake.so: undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)@GLIBCXX_3.4.26'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewStencilOpSeparate'
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_set_maxeval'
/opt/drake/lib/libdrake.so: undefined reference to `ClpModel::scaling(int)'
/opt/drake/lib/libdrake.so: undefined reference to `spdlog::pattern_formatter::pattern_formatter(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, spdlog::pattern_time_type, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/opt/drake/lib/libdrake.so: undefined reference to `std::filesystem::__cxx11::path::_List::end() const@GLIBCXX_3.4.26'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewUniformMatrix3fv'
/opt/drake/lib/libdrake.so: undefined reference to `vtable for Ipopt::OptionsList'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewDeleteRenderbuffers'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewTexImage3D'
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_set_min_objective'
/opt/drake/lib/libdrake.so: undefined reference to `std::filesystem::current_path[abi:cxx11]()@GLIBCXX_3.4.26'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewGetShaderiv'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewTransformFeedbackVaryings'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewBindFragDataLocation'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__GLEW_ARB_texture_float'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewDisableVertexAttribArray'
/opt/drake/lib/libdrake.so: undefined reference to `ClpSimplex::ClpSimplex(bool)'
/opt/drake/lib/libdrake.so: undefined reference to `ClpSimplex::~ClpSimplex()'
//opt/drake/lib/libdrake_ibex.so: undefined reference to `ClpSimplex::infeasibilityRay(bool) const'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__GLEW_ARB_transform_feedback3'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewBlitFramebuffer'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewBeginQuery'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_transfer_real@GFORTRAN_8'
//opt/drake/lib/libdrake_ibex.so: undefined reference to `ClpModel::setMaximumIterations(int)'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewFramebufferTexture2D'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_st_inquire@GFORTRAN_8'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewUniform1f'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewGenFramebuffers'
//opt/drake/lib/libdrake_ibex.so: undefined reference to `ClpSimplex::setColumnBounds(int, double, double)'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewClampColor'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_compare_string@GFORTRAN_8'
/opt/drake/lib/libdrake.so: undefined reference to `ClpSimplex::primal(int, int)'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewDeleteProgram'
[... 123 lines cut ...]
/opt/drake/lib/libdrake.so: undefined reference to `nlopt_get_dimension'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewUniform1iv'
/opt/drake/lib/libdrake.so: undefined reference to `std::filesystem::__cxx11::directory_iterator::operator++()@GLIBCXX_3.4.26'
/opt/drake/lib/libdrake.so: undefined reference to `ClpModel::loadQuadraticObjective(int, int const*, int const*, double const*)'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewCreateShader'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewGetStringi'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_transfer_logical_write@GFORTRAN_8'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewGenQueries'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewDetachShader'
/opt/drake/lib/libdrake.so: undefined reference to `_gfortran_string_index@GFORTRAN_8'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewGetTextureLevelParameteriv'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewStencilMaskSeparate'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewActiveTexture'
//opt/drake/lib/libvtkRenderingOpenGL2-9.1.so.1: undefined reference to `__glewDrawElementsInstancedARB'
[... cut ...]
src/find_resource/CMakeFiles/find_resource_example.dir/build.make:99: recipe for target 'src/find_resource/find_resource_example' failed
make[2]: *** [src/find_resource/find_resource_example] Error 1
CMakeFiles/Makefile2:1008: recipe for target 'src/find_resource/CMakeFiles/find_resource_example.dir/all' failed
make[1]: *** [src/find_resource/CMakeFiles/find_resource_example.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

How can I solve such error? I try to install drake on ubuntu18.04, I don't know whether that will influence my installation process? Thanks.

jwnimmer-tri, 2022-06-17:
What version of Drake are you using?  As of [v1.2.0](https://drake.mit.edu/release_notes/v1.2.0.html), Drake does not support Ubuntu 18.04 anymore.

Yingfan99327, 2022-06-17:
I guess I am trying to install drake of v1.3.0, so such error occurs mainly due to the version of the ubuntu system I use? I have to use ubuntu20.04, right?

jwnimmer-tri, 2022-06-17:
Yes; here are the supported versions:

https://drake.mit.edu/installation.html

closed 2024-09-05 by jwnimmer-tri