RobotLocomotion/drake-external-examples #233: The drake_cmake_installed example Particle does not plot in visualizer

huweiATgithub, 2022-07-24:
# How to reproduce
- Clone and change directory to [particle example](https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed/src/particles)
- mkdir build && cd build && cmake .. && make
- Terminal 1: `/opt/drake/bin/drake-visualizer`; Terminal 2: `./uniformly_accelerated_particle`

Then, the texts on the visualizer keep updating (sim time: xxx, real time factor: xxx). But I cannot find the particle. Also, from the Scene Browser of drake-visualizer, I see no scene of the added particle. (neither does the control pane in the meshcat)

# What did I try
- Replace DrakeVisualizer with MeshCatVisualizer.
- Add a simulator->Initialize(); and a sleep after it to look at the visualizer carefully.

# Thoughts
In class ParticleGeometry, the geometry has been added to the scene_graph with the illustration role assigned. Then, I think they will automatically be added to the visualizer attached later?

jwnimmer-tri, 2022-07-26:
Thanks for the report, we'll take a look.

The purpose of `drake-external-examples` is to show how to set up the _build system_ for your projects.  To see examples of Drake's features, a better place to look is drake.mit.edu/#tutorials or https://github.com/RobotLocomotion/drake/tree/master/examples/.

Of course the programs and demo instructions here should still work correctly, but in that light I will probably resolve this bug by deleting the particle code here and replacing it with something simpler.

huweiATgithub, 2022-07-28:
I might have found out the cause of this issue as raised in the drake repo: https://github.com/RobotLocomotion/drake/issues/17649

A current workaround to have things working properly is to drop usage of BuildInto.
I can submit a PR if you think this is ok.

jwnimmer-tri, 2024-09-05:
Thanks for reporting this!

The discussion in https://github.com/RobotLocomotion/drake/issues/17649 is more detailed at this point, so I'm going to close this issue as a duplicate of that other issue.  It's still a bug to be fixed, but it will be easier for the Drake team to track it using the other issue number.

PR #355 (2025-01-17): [drake_cmake_installed] Replace broken particles example
For now, just sync it up with the `drake_cmake_installed_apt` example. It is not a great demo, but it's what we have.

Closes #233.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake-external-examples/355)
closed 2024-09-05 by jwnimmer-tri