RobotLocomotion/drake-external-examples #249: Tutorials for getting up and running.

drewhamiltonasdf, 2022-12-16:
This package is in desperate need of basic tutorials for compiling hello world examples. The guzhaoyuan tutorials are out of date and don't compile, as far as I can tell. It requires a great deal of fiddling to get even the most basic examples running. The bazel build system is not explained well at all, and I found that I had to use cmake to get anything to run.

I know it's a headache mantaining open-source projects, but some basic documentation will increase user numbers greatly. If it takes several days just to get started, most people will return back to their familiar tools, and always remember Drake as the thing they couldn't get to work.

That said, I personally am committed to the project of getting things running because Drake does offer so many powerful tools for modern control. So hats off to everyone that contributes.

jwnimmer-tri, 2023-01-03:
We have https://drake.mit.edu/#tutorials about the robotics features, but I think you're asking about a tutorial for how to compile your own software that uses Drake as a dependency?

For starters, for new users I recommend using Python and https://drake.mit.edu/pip.html.  It's very easy to get started that way.

For users who need to code in C++, the goal of this repository is that the user can take something like https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed_apt, copy those files into their own project as a "starter kit" with almost no modifications, and then be up and running almost immediately.

Was that intention clear?  (i.e., did you understand that the top-level directories in this repository were each their own standalone tutorial)?

If yes, then which one(s) of the examples did you try, and could you please explain in more detail any difficulties you had with them?

jwnimmer-tri, 2024-09-05:
Closing for lack of response.  If I can't understand the problem, I can't try to improve the status quo.

closed 2024-09-05 by jwnimmer-tri