RobotLocomotion/drake-external-examples #266: Example for PyPI workflow

jwnimmer-tri, 2023-04-06:
All of our examples show complicated C++ things like CMake, Bazel, Ament, etc. The majority of our new users are Python users though!

We should have an example showing the PyPI workflow.  If nothing else, a README linking to https://drake.mit.edu/pip.html would be a starting point.  We could probably show an example using poetry or something though.

\CC @rcory FYI

jwnimmer-tri, 2025-01-06:
I guess we need to choose a name for the new example.  Maybe `drake-pypi`?

Also maybe we should try making it a PDM project, instead of Poerty?  Any advice @mwoehlke-kitware?

mwoehlke-kitware, 2025-01-06:
> Also maybe we should try making it a PDM project, instead of Poerty?

In my limited experience with both, they seem to be fairly similar. That said, Poetry seems to work well as a front-end (e.g. `poetry run ...` Just Works™) whereas, while I _think_ PDM has something similar, I haven't used it.

That said, Drake is moving toward PDM for tarball distributions, but since I presume we're talking about wheels here, I believe PDM is not involved on the user end.

I think for this case, it's more important to try to reach the most users. That might be Poetry, might be PDM, or might be `python -m venv` + `pip install`. (Or it might be *more than one* of those options.)

jwnimmer-tri, 2025-01-06:
Good points.  I think my first goal here is to show users something that will feel familiar, so maybe we should do `drake-poetry` as the new example, with a demo project that uses Poetry to consume drake from PyPI.

Later (different ticket), we could add others like `drake-pdm` or `drake-basic-pip` if necessary.

mwoehlke-kitware, 2025-01-06:
If we have a good sense of which is most useful to users, we should do that; I don't think it matters what we're doing internally.

That said, I asked KW's internal Python channel, and someone made the good point that `python -m venv` + `pip install` is probably the easiest for beginners, and people wanting to use Poetry/PDM are likely less in need of hand-holding.

jwnimmer-tri, 2025-01-08:
Okay, I guess both a basic pip example and a poetry example are probably worth doing.  I'm okay with whatever order y'all want to attack those in.

jwnimmer-tri, 2025-01-13:
BTW we can test these using only GHA (not Jenkins).  Also IIRC there is a GHA "step" available for pip caching.

rcory, 2025-01-14:
@jwnimmer-tri I don't think you meant to tag me in your top level comment.

jwnimmer-tri, 2025-01-14:
Yes, I did (in 2023).  You asked about this w.r.t. helping out your university collaborators.

rcory, 2025-01-14:
I'd be lying if I said I remembered that. But, I believe you.

jwnimmer-tri, 2025-01-27:
Fixed in #358.

closed 2025-01-27 by jwnimmer-tri