RobotLocomotion/drake-external-examples #273: Where are the executables using drake_cmake_external??

xiao-wang007, 2023-10-31:
Hi 
This may be a silly question, but I cannot find the executables after build in cmake. 
In addition, where is the executable of the `simple_continuouse_time_system.cc`??
If using drake as external, do I just need to put everything under `app/`

Cheers

jwnimmer-tri, 2023-10-31:
The README says to build it like this:

> To build all apps using CMake and Make:
>
> ```
> mkdir build
> cd build
> cmake ..
> make
> ```

Assuming you've done that, the binary is at `./drake_external_examples/apps/simple_continuous_time_system` (in other words, `drake-external-examples/drake_cmake_external/build/drake_external_examples/apps/simple_continuous_time_system`).

>  If using drake as external, do I just need to put everything under `app/`?

No.  You can use any directory names you want.

xiao-wang007, 2023-10-31:
Great! Thanks.

xiao-wang007, 2023-10-31:
Great, thanks!

closed 2023-10-31 by xiao-wang007