RobotLocomotion/drake-external-examples #291: Build is not successful using drake_cmake_external

xiao-wang007, 2024-06-06:
Hi 

Just trying to code using drake c++ as external in cmake. The build produces this error. I have not added any code, just the first build after git clone.

Cheers! 

server_log: /Users/xiao/0_codes/drake_cmake_external/build/drake/_bazel_xiao/a7c5cbe7e7dda40ca5ecbbe8e4a7355b/java.log.xiao-m2.xiao.log.java.20240606-093636.63778
server_pid: 63778
used-heap-size: 83MB
workspace: /Users/xiao/0_codes/drake_cmake_external/build/drake/drake_build_cwd
-- Configuring done (2.8s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/xiao/0_codes/drake_cmake_external/build/drake
[ 75%] Performing build step for 'drake'
Built target drake_version
INFO: Analyzed target //:install (317 packages loaded, 26048 targets configured).
INFO: Found 1 target...
ERROR: /Users/xiao/0_codes/drake_cmake_external/build/drake-prefix/src/drake/tools/install/libdrake/BUILD.bazel:48:11: RunBinary tools/install/libdrake/stamp_version.cmake failed: (Exit 1): parse_version failed: error executing RunBinary command (from target //tools/install/libdrake:drake_parse_version_stamp) bazel-out/darwin_arm64-opt-exec-ST-13d3ddad9198/bin/tools/install/libdrake/parse_version bazel-out/darwin_arm64-opt/bin/tools/install/libdrake/stamp_version.txt ... (remaining 1 argument skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
Traceback (most recent call last):
  File "/Users/xiao/0_codes/drake_cmake_external/build/drake/_bazel_xiao/a7c5cbe7e7dda40ca5ecbbe8e4a7355b/sandbox/darwin-sandbox/26/execroot/drake/bazel-out/darwin_arm64-opt-exec-ST-13d3ddad9198/bin/tools/install/libdrake/parse_version.runfiles/drake/tools/install/libdrake/parse_version.py", line 96, in <module>
    main()
  File "/Users/xiao/0_codes/drake_cmake_external/build/drake/_bazel_xiao/a7c5cbe7e7dda40ca5ecbbe8e4a7355b/sandbox/darwin-sandbox/26/execroot/drake/bazel-out/darwin_arm64-opt-exec-ST-13d3ddad9198/bin/tools/install/libdrake/parse_version.runfiles/drake/tools/install/libdrake/parse_version.py", line 90, in main
    _write_version_info(args.output, *_parse_stamp(args.input))
                                      ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/xiao/0_codes/drake_cmake_external/build/drake/_bazel_xiao/a7c5cbe7e7dda40ca5ecbbe8e4a7355b/sandbox/darwin-sandbox/26/execroot/drake/bazel-out/darwin_arm64-opt-exec-ST-13d3ddad9198/bin/tools/install/libdrake/parse_version.runfiles/drake/tools/install/libdrake/parse_version.py", line 40, in _parse_stamp
    raise ValueError(f'Version {version_full} is not valid')
ValueError: Version 0.0.20240606.093638+unknown is not valid
Target //:install failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 23.228s, Critical Path: 0.95s
INFO: 592 processes: 567 internal, 25 darwin-sandbox.
ERROR: Build did NOT complete successfully
make[5]: *** [CMakeFiles/drake_cxx_python] Error 1
make[4]: *** [CMakeFiles/drake_cxx_python.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [drake-prefix/src/drake-stamp/drake-build] Error 2
make[1]: *** [CMakeFiles/drake.dir/all] Error 2
make: *** [all] Error 2

jwnimmer-tri, 2024-06-06:
+@mwoehlke-kitware can you investigate, please?

It looks to me like some logic for HHMMSS ends creating `093638` which I guess isn't PEP 440 compliant (leading zero)?

mwoehlke-kitware, 2024-06-06:
Ah... does this only affect builds where CI is not setting the version? There is logic in drake-ci to strip leading zeros. Looks like Drake `generate_version.cmake` should have the same logic.

mwoehlke-kitware, 2024-06-06:
@xiao-wang007, thanks for the report!

Was this using Drake that you built yourself (i.e. not a downloaded package)? If so, I think https://github.com/RobotLocomotion/drake/pull/21532 will fix it. Note, however, that the problem only occurs if you build Drake before 10:00. (I believe that's local time, not UTC.)

jwnimmer-tri, 2024-06-06:
Yup, we'll hypothesize that https://github.com/RobotLocomotion/drake/pull/21532 was the fix.  We can re-open if we see more evidence of failures.

closed 2024-06-06 by jwnimmer-tri