RobotLocomotion/drake-external-examples #302: Segment Fault When Run Drake_CMake_Installed simple_bindings example

hanlanqian, 2024-09-15:
Hey, I'm new to using Drake and Pybind11. I just followed the README file and tried to test the simple_bindings example. However, I encountered a segmentation fault when running the command `adder = builder.AddSystem(SimpleAdder(100.))`. It seems like there's an issue when calling the C++ Drake function in Python with Pybind11 and I can't figure out the reason.
Following is the cmake output:
```
[main] Configuring project: drake_cmake_installed 
[proc] Executing command: /usr/bin/cmake -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/gcc -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/g++ --no-warn-unused-cli -S/home/${username}/projects/drake-external-examples/drake_cmake_installed -B/home/${username}/projects/drake-external-examples/drake_cmake_installed/build -G "Unix Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Found pybind11: /usr/include (found version "2.9.1")
[cmake] -- Configuring done
[cmake] You have changed variables that require your cache to be deleted.
[cmake] Configure will be re-run and you may have to reset some variables.
[cmake] The following variables have changed:
[cmake] CMAKE_C_COMPILER= /usr/bin/gcc
[cmake] CMAKE_CXX_COMPILER= /usr/bin/g++
[cmake] 
[cmake] -- The C compiler identification is GNU 11.4.0
[cmake] -- The CXX compiler identification is GNU 11.4.0
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: /usr/bin/gcc - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: /usr/bin/g++ - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Looking for pthread.h
[cmake] -- Looking for pthread.h - found
[cmake] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[cmake] -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
[cmake] -- Found Threads: TRUE  
[cmake] -- Found Python3: /home/${username}/miniconda3/envs/drake/bin/python (found version "3.10.14") found components: Interpreter Development Development.Module Development.Embed 
[cmake] -- Performing Test HAS_FLTO
[cmake] -- Performing Test HAS_FLTO - Success
[cmake] -- Found pybind11: /usr/include (found version "2.9.1")
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: /home/${username}/projects/drake-external-examples/drake_cmake_installed/build
[cpptools] The build configurations generated do not contain the active build configuration. Using "" for CMAKE_BUILD_TYPE instead of "Debug" to ensure that IntelliSense configurations can be found
```
Failed Test Logfile:
```
5/6 Testing: simple_bindings_test
5/6 Test: simple_bindings_test
Command: "/home/elaina/miniconda3/envs/drake/bin/python" "/home/elaina/projects/drake-external-examples/drake_cmake_installed/src/simple_bindings/simple_bindings_test.py"
Directory: /home/elaina/projects/drake-external-examples/drake_cmake_installed/build/src/simple_bindings
"simple_bindings_test" start time: Sep 15 10:42 CST
Output:
----------------------------------------------------------
<end of output>
Test time =   0.12 sec
----------------------------------------------------------
Test Failed.
"simple_bindings_test" end time: Sep 15 10:42 CST
"simple_bindings_test" time elapsed: 00:00:00
----------------------------------------------------------
```

jwnimmer-tri, 2024-09-25:
Thanks for reporting and asking!

I am not 100% sure about the problem, but I do see one suspicious thing in the log:
```
Found pybind11: /usr/include (found version "2.9.1")
```

The example is supposed to be finding the custom version of pybind11 available at `drake/lib/cmake/pybind11/...` , because the build command in the README specifies `-DCMAKE_PREFIX_PATH=$HOME/drake`.  Using a non-Drake version of pybind11 will probably not work, and indeed fail with segfaults (which seems to match what you are seeing).

We usually don't test this example with a version of pybind11 installed system-wide in `/usr/include`, so it's possible that the order of precedence with CMake finding pybind11 has a mistake in it.

I can have our team try to reproduce this problem, but first can you quickly confirm -- did you do these exact steps from the README ...

```md
###############################################################
# Build Everything
###############################################################
git clone https://github.com/RobotLocomotion/drake-external-examples.git
cd drake-external-examples
mkdir drake_cmake_installed-build && cd drake_cmake_installed-build
cmake -DCMAKE_PREFIX_PATH=$HOME/drake ../drake_cmake_installed
make
```

... and is `$HOME/drake` where the `tar` command unpacked Drake into?

hanlanqian, 2024-09-29:
Thanks for your reply and helpful explanation! This error does source from the linking to customed install pybind11.

closed 2024-09-29 by hanlanqian