RobotLocomotion/drake-external-examples #303: Run each Jenkins build flavor on its on machine

BetsyMcPhail, 2024-09-16:
In #300, all but 2 jobs were removed from Jenkins because they also existed in GHA. In this issue, the task is to update the Jenkinsfile so that the remaining two jobs (drake_bazel_external and drake_cmake_external), each run on their own node. This way we can ensure that the setup scripts and results remain independent.

Follow up to #61

BetsyMcPhail, 2024-11-13:
As part of #322 each job runs on a fresh instance. However, there is still one Jenkinsfile, that should be split up, see #309.

jwnimmer-tri, 2024-11-14:
A single `Jenkinsfile` is all we want / need.  The "self-contained examples" goal does not encompass showing users how to configure Jenkins -- only GHA.

BetsyMcPhail, 2024-11-14:
Great, then I think we can call this one complete?

closed 2024-11-14 by jwnimmer-tri