RobotLocomotion/drake-external-examples #310: Broken link in ament cmake installed README

BetsyMcPhail, 2024-09-19:
In the drake_ament_cmake_installed [README](https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_ament_cmake_installed#readme), the `ament_cmake` [link](https://github.com/RobotLocomotion/drake-external-examples/blob/70a3831a18c3fe7e4bffcae1ef785a97629b583e/drake_ament_cmake_installed/README.md?plain=1#L3) is broken.

ahcorde, 2024-10-30:
The right link is https://docs.ros.org/en/rolling/How-To-Guides/Ament-CMake-Documentation.html

you might want to change the ROS distro branch

jwnimmer-tri, 2024-10-30:
Great, thanks!

Since the example is hard-coded to Humble, let's use this link:
- https://docs.ros.org/en/humble/How-To-Guides/Ament-CMake-Documentation.html

When we upgrade from Humble to something newer, we can change the docs and setup scripts at the same time.

+@BetsyMcPhail could you or @nicolecheetham please open a PR for this?

jwnimmer-tri, 2024-10-31:
Fixed in #337.

closed 2024-10-31 by jwnimmer-tri