RobotLocomotion/drake-external-examples #329: Skipped catkin tests

nicolecheetham, 2024-10-21:
Currently, the catkin installed job successfully runs however it skips all of its tests.

Agenda:
1. Check if other examples' tests are skipped
2. Investigate whether the job(s) always had skipped tests or if the DEE refactoring caused the tests to be skipped

nicolecheetham, 2024-10-23:
PR 328 first brought this issue up.

nicolecheetham, 2024-10-23:
* ament_cmake_installed runs a test executable (does not mention specific tests when run)
* catkin_installed does not run tests 

The rest of the examples explicitly mention tests in the build and test stage.

jwnimmer-tri, 2024-10-23:
One easy way to cross-check would be to introduce a failure or bug into one of the test cases, and confirm CI shows red.

nicolecheetham, 2024-10-23:
I created a branch with the most recent commit from main being "Add some missing semicolons PR 294" that was commited Jun 25. For context, I made my first commit to this repo Aug 21st.

The catkin installed job had the same warning as what was introduced in PR 328. As such, the catkin tests that require gtest were already skipped when the refactoring started taking place.

Here is the warning below:
```
CMake Warning at /usr/share/catkin/cmake/test/gtest.cmake:159 (message):
  skipping gtest 'drake_catkin_installed_test' in project
  'drake_catkin_installed' because gtest was not found
Call Stack (most recent call first):
  /usr/share/catkin/cmake/test/gtest.cmake:89 (_catkin_add_executable_with_google_test)
  /usr/share/catkin/cmake/test/gtest.cmake:37 (_catkin_add_google_test)
  drake_catkin_installed/CMakeLists.txt:[54](https://github.com/RobotLocomotion/drake-external-examples/actions/runs/11486269912/job/31968186715?pr=330#step:5:55) (catkin_add_gtest)
```

jwnimmer-tri, 2024-10-29:
Let's do the following instead -- open two new issues that are more crisp, and then close this issue as a duplicate of those two.

The two new issues should be titled:
* [drake_ament_cmake_installed] Failing tests do not show up in CI
* [drake_catkin_installed] Failing tests do not show up in CI

Each ticket should provide a recipe and/or evidence that a broken test still has GHA output showing up as a green bubble, whereas we want the red x in the PR status.  Essentially, make them a good bug report that other people can work on.

Then, I will assign two issues to the ROS people who help curate this repository, and ask them to fix it (not Kitware).

nicolecheetham, 2024-10-29:
@jwnimmer-tri Okay, will do

nicolecheetham, 2024-10-30:
Closing this issue as it is now a duplicate of two newly created issues.

nicolecheetham, 2024-10-30:
Closing this issue as it is now a duplicate of two newly created issues

closed 2024-10-30 by nicolecheetham