RobotLocomotion/drake-external-examples #351: Run Jenkins jobs in parallel

jwnimmer-tri, 2025-01-09:
As shown by our [Jenkins timeline](https://drake-jenkins.csail.mit.edu/job/RobotLocomotion/job/drake-external-examples/job/main/), our [Jenkinsfile](https://github.com/RobotLocomotion/drake-external-examples/blob/main/Jenkinsfile) runs the three flavors of examples one after another.  With our recent work to split up the examples, the three flavors are now fully independent, so it would be helpful to run them in parallel for faster PR turn-around latency.  After some months we'll be removing the "legacy" flavor so it won't be quite to bad, but having a 25-minute turn instead of a 50-minute turn would still be useful.

BetsyMcPhail, 2025-01-21:
While testing this issue, I noticed that Jenkins jobs for PRs get re-run when a PR is merged. 

For example, #349 hasn't had any changes since 1/10 _but_ Jenkins jobs ran on 1/16 and 1/17 due to "Branch Indexing" (https://drake-jenkins.csail.mit.edu/job/RobotLocomotion/job/drake-external-examples/job/PR-349/). PRs #352 and #355 were merged on 1/16 and 1/17.

I think we just need to uncheck "Scan Repository Triggers" / "Periodically  if not otherwise run" / "1 hour" to disable this behavior.

Before I did that, @jwnimmer-tri I wanted to double-check that this is _not_ desired behavior.

jwnimmer-tri, 2025-01-21:
I could go either way, but we might want to keep it.  The purpose of that setting is to make sure that PR job results don't get too stale, which helps prevent broken code from landing on master in case some other PR landed after the most recent build.

Certainly "1 hour" is too fast, though.  Something like "1 day" or even "7 days" would be more appropriate.

BetsyMcPhail, 2025-01-21:
Ah, okay, that makes sense. I'm going to update it to 1 day for now.

BetsyMcPhail, 2025-01-21:
Nevermind, the configuration isn't editable from the UI and I don't see it explicitly set in the Jenkinsfile, so I'm happy to call it good enough for now.

PR #360 (2025-01-27): Run Jenkins jobs in parallel
Resolves #351

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake-external-examples/360)
closed 2025-01-27 by BetsyMcPhail