RobotLocomotion/drake-external-examples #359: Remove sudo from READMEs

jwnimmer-tri, 2025-01-17:
For any scripts in this repository, if they require root access then they should internally obtain that access (e.g., by conditionally prefixing certain commands like `apt-get` with a `sudo`).  They should _not_ instruct the user to run `sudo setup/install_prereqs`.

This is the policy Drake will be aiming for, moving forward (see e.g. https://github.com/RobotLocomotion/drake/pull/22461).

jwnimmer-tri, 2025-01-17:
I noticed this in https://github.com/RobotLocomotion/drake-external-examples/pull/357.  It's OK by me if you'd like to merge the correct documentation first (which currently means telling the user to run sudo) and then circle back here later.

(However, that might be difficult in case macOS requires no-sudo but Ubuntu requires yes-sudo, and they share the same README.)

BetsyMcPhail, 2025-01-17:
I was just reviewing this on #357. I believe the `setup/install_prereqs` scripts in this repo (e.g. https://github.com/RobotLocomotion/drake-external-examples/blob/main/drake_bazel_external/setup/install_prereqs) are already setup as you described. Can you please confirm this is what you're looking for?

jwnimmer-tri, 2025-01-17:
That's the right idea, we could call it 90% of the way there.  The problem is that if someone is running in Docker and they are already root, sometimes `sudo` actually _fails_.

The ideal would be to only run `sudo` if we're not already root.  Here's how I did that in my recent docs PR:

https://github.com/RobotLocomotion/drake/pull/22461/commits/52876ee9f1f2666e49d727c8d67909b788030f90#diff-4ab58ad02a221a36b723ddbb13f04ef076ad887fa01c1d6fd0b1809ed9b39d1eR26-R30

BetsyMcPhail, 2025-02-11:
Fixe in #362

closed 2025-02-11 by BetsyMcPhail