RobotLocomotion/drake-external-examples #368: [drake_cmake_installed] CI failing on venv

jwnimmer-tri, 2025-02-08:
See https://github.com/RobotLocomotion/drake-external-examples/actions/runs/13215389169/job/36894041170 and prior.

This might be an upstream Drake issue, or a broken GHA CI issue, not sure yet:

```console
+ venv_root=/Users/runner/drake
+ python3.12 -m venv /Users/runner/drake
+ /Users/runner/drake/bin/pip3 install -U -r /Users/runner/drake/share/drake/setup/requirements.txt
...
Installing collected packages: distlib, typing-extensions, truststore, tomlkit, socksio, sniffio, shellingham, resolvelib, python-dotenv, pyproject-hooks, pygments, platformdirs, pbs-installer, packaging, msgpack, mdurl, installer, idna, h11, filelock, certifi, blinker, virtualenv, markdown-it-py, httpcore, findpython, dep-logic, anyio, rich, httpx, unearth, hishel, pdm
Successfully installed anyio-4.8.0 blinker-1.9.0 certifi-2024.12.14 dep-logic-0.4.10 distlib-0.3.9 filelock-3.17.0 findpython-0.6.2 h11-0.14.0 hishel-0.1.1 httpcore-1.0.7 httpx-0.28.1 idna-3.10 installer-0.7.0 markdown-it-py-3.0.0 mdurl-0.1.2 msgpack-1.1.0 packaging-24.2 pbs-installer-2025.1.6 pdm-2.22.3 platformdirs-4.3.6 pygments-2.19.1 pyproject-hooks-1.2.0 python-dotenv-1.0.1 resolvelib-1.1.0 rich-13.9.4 shellingham-1.5.4 sniffio-1.3.1 socksio-1.0.0 tomlkit-0.13.2 truststore-0.10.0 typing-extensions-4.12.2 unearth-0.17.2 virtualenv-20.29.1
+ /Users/runner/drake/bin/pdm use -p /Users/runner/drake/share/drake/setup -f /Users/runner/drake
INFO: Using the first matched interpreter.
Using Python interpreter: /Users/runner/drake/bin/python3 (3.12)
+ /Users/runner/drake/bin/pdm sync -p /Users/runner/drake/share/drake/setup --prod
STATUS: Resolving packages from lockfile...
...
See /Users/runner/Library/Logs/pdm/pdm-install-8qi4ijxr.log for detailed debug log.
  ✖ Install widgetsnbextension 4.0.13 failed
[InstallationError]: Some package operations failed.
...
 line 110, in <module>
    urllib3.__version__, chardet_version, charset_normalizer_version
    ^^^^^^^^^^^^^^^^^^^
AttributeError: module 'urllib3' has no attribute '__version__'
...
```

jwnimmer-tri, 2025-02-08:
This looks like the problem to me:

```
+ /Users/runner/drake/bin/pdm use -p /Users/runner/drake/share/drake/setup -f /Users/runner/drake
INFO: Using the first matched interpreter.
Using Python interpreter: /Users/runner/drake/bin/python3 (3.12)
```

I imagine that this line is missing a flag to tell it to use 3.12?

https://github.com/RobotLocomotion/drake/blob/7713cb3983680bd8773f3826e56b0bc32bfd20c5/setup/mac/binary_distribution/install_prereqs.sh#L74

mwoehlke-kitware, 2025-02-10:
> I imagine that this line is missing a flag to tell it to use 3.12?

That's *sort of* what `pdm use` is doing; telling it to use the venv at `${venv_root}`. The "first matched interpreter" in the Drake venv is going to be the right one. I think there's a way to tell it what version to use explicitly, but that requires `install_prereqs.sh` to know the version, somehow, and practically speaking I don't think that gains us anything except suppressing the INFO message.

jwnimmer-tri, 2025-02-10:
Okay, my diagnosis was off the mark.  Still, there is some kind of bug here.

jwnimmer-tri, 2025-02-10:
Maybe it's a race condition w.r.t the order things are installed?  The build has been green since those two instances.

mwoehlke-kitware, 2025-02-10:
This doesn't seem to be failing at present?

The first error(s) are:
```
  ✖ Install websocket-client 1.8.0 failed
  ✖ Install widgetsnbextension 4.0.13 failed
  ...
  ✖ Install websocket-client 1.8.0 failed
See /Users/runner/Library/Logs/pdm/pdm-install-8qi4ijxr.log for detailed debug log.
  ✖ Install widgetsnbextension 4.0.13 failed
```

As with compiler errors, I'm suspicious of the reliability of anything after that, and failed installs that are working now suggest either a network issue or something funky on the PyPI end... which seems to have resolved itself in the mean time. For now at least, I don't know that there is anything to do, here, except maybe record the incident.

For CI, we might want to add something to try to dump that log file if it exists?

jwnimmer-tri, 2025-02-10:
It seems to me (especially in [this one](https://github.com/RobotLocomotion/drake-external-examples/actions/runs/13208674696/job/36877714623)) like pdm is calling `requests` (which uses `urllib3`) during the install process to download stuff, but at the same time is upgrading _it's own copy_ of `requests` in the venv out from under itself, leading to chaos depending on the order in which things get installed (which perhaps depends on internet weather).  This might not repro locally with a warm download cache.

It seems the use is 'pdm' calls 'unearth' which has a try-import of requests:
https://github.com/frostming/unearth/blob/fbf4165b2a22bccddd2e865cfd8d614674286605/src/unearth/finder.py#L50

When this happens in the middle of installing `requests` and `urllib3`, things fail because the try-import only catches `ImportError` not "the package was only halfway installed error".

jwnimmer-tri, 2025-02-10:
Possibly the fastest fix would be to force `requests` into the `requirements.txt` so that it's already installed by the time pdm starts up?

mwoehlke-kitware, 2025-02-10:
Eww. Okay, that smells like a PDM bug... although I'd love to "fix" it by getting jupyter off of requests...

Alternatively, I wonder if we should do what Drake itself does and have PDM in its own separate venv?

jwnimmer-tri, 2025-02-10:
Assuming it's a real bug, then it's a regression and we either fix forward something in the new few days before the release (yay), or else we revert the binary release venv PR and try again after the next release (yuck).

> Alternatively, I wonder if we should do what Drake itself does and have PDM in its own separate venv?

This seems like it's probably safer long term, but might be more effort than we have time for this week.  I'm okay with a hotfix now and a more involved refactor afterwards, if you like.

jwnimmer-tri, 2025-02-11:
Likely fixed by https://github.com/RobotLocomotion/drake/pull/22607.

closed 2025-02-11 by jwnimmer-tri