RobotLocomotion/drake-external-examples #378: Upgrade `drake-external-examples/drake_poetry` to Poetry 2.0 standards

tyler-yankee, 2025-03-18:
The `drake_poetry` example was created right around the time that Poetry released their 2.0, with a set of new standards for how `pyproject.toml` files should be structured. Our current `pyproject.toml` file is fine, but the `dependabot` can't parse it to bump the supported version of Drake when a new release comes out. And besides, we  should keep things up to date for Poetry/Drake end users.

* Update the `pyproject.toml` to be formatted according to the new standards (notably, should use the `[project]` table).
* Update the supported version of Drake to match the convention in `drake_pip`. This may require some research, but it seems that [Poetry now supports `~=`](https://python-poetry.org/docs/dependency-specification/), so we don't have to use `^`.
* Update `dependabot.yml` as necessary to bump `pyproject.toml`. We may also be able to use the [`versioning-strategy` field](https://docs.github.com/en/code-security/dependabot/working-with-dependabot/dependabot-options-reference#versioning-strategy--).

jwnimmer-tri, 2025-03-18:
(FYI I transferred this issue over from the Drake repo into this repo.)

tyler-yankee, 2025-04-02:
Closed as completed by #379.

closed 2025-04-02 by tyler-yankee