RobotLocomotion/drake-external-examples #381: Build error when using drake as cmake external

xiao-wang007, 2025-04-13:
Hi 
I got the following error when building drake in cmake as external.

OS: macos 
chip: apple chip

The only changes I made in the CMakeLists.txt under drake_cmake_external are the latest versions of fmt and spd log. Because with the default version, it gives me error saying cmake < 3.5 is not supported anymore.
Have I done anything wrong?

-----------------------------------------------------------------------------------------------------------------------------------------------
ERROR: /Users/xiao/0_codes/drake-external-examples/drake_cmake_external/build/drake/_bazel_xiao/6e05939919dd6d766bdf4415ba4d40a2/external/drake+/systems/controllers/BUILD.bazel:127:17: Compiling systems/controllers/pid_controlled_system.cc failed: (Exit 1): wrapped_clang_pp failed: error executing CppCompile command (from target @@drake+//systems/controllers:pid_controlled_system) external/apple_support++apple_cc_configure_extension+local_config_apple_cc/wrapped_clang_pp '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign ... (remaining 166 arguments skipped)

Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
external/drake+/systems/controllers/pid_controlled_system.cc:73:29: error: a template argument list is expected after a name prefixed by the template keyword [-Wmissing-template-arg-list-after-template-kw]
   73 |   plant_ = builder.template AddSystem(std::move(plant));
      |                             ^
1 error generated.

jwnimmer-tri, 2025-04-13:
Which version of XCode / AppleClang are you using?

tyler-yankee, 2025-04-14:
> the latest versions of fmt and spd log ... cmake < 3.5 is not supported anymore

This won't solve the error you're seeing, but we can update `fmt` and `spdlog` to their most recent versions so that you and other users don't run into this. Compatibility with versions of CMake < 3.5 has been [removed in CMake 4.0](https://cmake.org/cmake/help/latest/release/4.0.html#deprecated-and-removed-features).

jwnimmer-tri, 2025-04-14:
> However, note the versions of CMake officially supported by Drake ...

From the brew output on our new AWS images, it looks like CMake in our macOS CI is 4.0.1 now, so we should update our docs to say (only) 4.0 is supported on macOS, to match the images.

closed 2025-08-22 by jwnimmer-tri