RobotLocomotion/drake-external-examples #393: Fix PyPI dependabot Drake upgrades

tyler-yankee, 2025-05-19:
We tried to set up dependabot to automatically bump Drake's version in #375, but something is up with it. After releasing 1.41.0, #392 was opened which incorrectly changes 1.40 -> 1.31 (??), and drake_poetry is not detected at all. Something similar has happened with previous releases.

mwoehlke-kitware, 2025-05-19:
> Not picking up on drake_poetry may be related to the incidental change introduced in #382 -- `~` to `~=`.

Unlikely. That change is correct; not only does `~=` have the desired meaning, it's not clear that `~` is even valid in that context. (Note: the specifier was originally `^x.y.z`, which is equivalent to `>= x.y.z, ~= x.y`. We lost the ability to require a minimum point release, but OTOH both pip and Poetry now have the exact same specifier. :shrug:)

jwnimmer-tri, 2025-05-27:
That sounds good to me.  Please just have 1 other Kitware person cross-check your work on trying dependabot, and if you both agree that it's terrible, then feel free to go ahead and switch DEE to renovate.

tyler-yankee, 2025-05-27:
That makes sense to me, we had a brief discussion about it yesterday.

I also tried adding `versioning-strategy: increase` to the config to no avail locally, and I doubt that would solve both seemingly unrelated issues with pip and poetry anyway.

#394 is pending review to disable dependabot for the two examples.

closed 2025-06-25 by jwnimmer-tri